stylesuxx/esc-configurator #424: having a proplem configrating the esc "Trying to read 0 ESC"

shamssh123, 2023-11-18:
dont know why but i cant connect my port to the esc configrator 
i did connected it ones but it didnt work for me 
after that it shows me this massge

2023/11/18 01:30:27: MSP command failed: Timed out after 1000ms
2023/11/18 01:31:11: Chrome (119.0.0.0) on Windows
2023/11/18 01:31:11: Plugged in
2023/11/18 01:31:13: Opened serial port
2023/11/18 01:31:14: MSP command failed: Timed out after 1000ms
2023/11/18 01:31:16: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 01:31:17: MSP command failed: Timed out after 1000ms
2023/11/18 01:31:17: Port already in use by another application - try re-connecting
2023/11/18 01:31:38: Opened serial port
2023/11/18 01:31:39: MSP command failed: Timed out after 1000ms
2023/11/18 01:31:42: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 01:31:43: MSP command failed: Timed out after 1000ms
2023/11/18 01:31:43: Port already in use by another application - try re-connecting
2023/11/18 01:34:28: Unplugged
2023/11/18 01:34:48: NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user. 
2023/11/18 01:35:56: NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user. 
2023/11/18 01:36:08: NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user. 
2023/11/18 01:36:55: Plugged in
2023/11/18 01:37:02: Unplugged
2023/11/18 01:37:19: Plugged in
2023/11/18 01:37:41: Opened serial port
2023/11/18 01:37:41: MultiWii API version received: 1.45.0
2023/11/18 01:37:41: Flight controller info, identifier: BTFL version: 4.4.3
2023/11/18 01:37:41: Running firmware released on: Nov 14 2023 16:10:10
2023/11/18 01:37:41: Board: S405, version: 0
2023/11/18 01:37:41: Unique device ID received - 0x03f002c3033470f37393338
2023/11/18 01:37:47: Trying to read 0 ESCs
2023/11/18 01:38:00: MSP command failed: Timed out after 1000ms
2023/11/18 01:42:46: Chrome (119.0.0.0) on Windows
2023/11/18 01:42:46: Plugged in
2023/11/18 01:43:09: Port selected
2023/11/18 01:43:12: Opened serial port
2023/11/18 01:43:13: MSP command failed: Timed out after 1000ms
2023/11/18 01:43:15: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 01:43:16: MSP command failed: Timed out after 1000ms
2023/11/18 01:43:16: Port already in use by another application - try re-connecting
2023/11/18 01:44:43: Unplugged
2023/11/18 01:45:09: Plugged in
2023/11/18 01:45:11: Opened serial port
2023/11/18 01:45:11: MultiWii API version received: 1.45.0
2023/11/18 01:45:11: Flight controller info, identifier: BTFL version: 4.4.3
2023/11/18 01:45:11: Running firmware released on: Nov 14 2023 16:10:10
2023/11/18 01:45:11: Board: S405, version: 0
2023/11/18 01:45:11: Unique device ID received - 0x03f002c3033470f37393338
2023/11/18 01:45:16: Trying to read 0 ESCs
2023/11/18 01:46:18: MSP command failed: Timed out after 1000ms
2023/11/18 01:50:54: Reader failed NetworkError: The device has been lost.
2023/11/18 01:50:54: Unplugged
2023/11/18 01:50:56: Plugged in
2023/11/18 01:53:14: Unplugged
2023/11/18 01:55:03: Plugged in
2023/11/18 01:55:30: Port selected
2023/11/18 01:55:33: Chrome (119.0.0.0) on Windows
2023/11/18 01:55:33: Plugged in
2023/11/18 01:55:35: Opened serial port
2023/11/18 01:55:35: MultiWii API version received: 1.45.0
2023/11/18 01:55:36: Flight controller info, identifier: BTFL version: 4.4.3
2023/11/18 01:55:36: Running firmware released on: Nov 14 2023 16:10:10
2023/11/18 01:55:36: Board: S405, version: 0
2023/11/18 01:55:36: Unique device ID received - 0x03f002c3033470f37393338
2023/11/18 01:55:40: Closed port
2023/11/18 01:55:47: Opened serial port
2023/11/18 01:55:47: MultiWii API version received: 1.45.0
2023/11/18 01:55:47: Flight controller info, identifier: BTFL version: 4.4.3
2023/11/18 01:55:47: Running firmware released on: Nov 14 2023 16:10:10
2023/11/18 01:55:47: Board: S405, version: 0
2023/11/18 01:55:47: Unique device ID received - 0x03f002c3033470f37393338
2023/11/18 01:55:50: Trying to read 0 ESCs
2023/11/18 01:55:54: MSP command failed: Timed out after 1000ms
2023/11/18 01:55:58: Reader failed NetworkError: The device has been lost.
2023/11/18 01:56:14: Chrome (119.0.0.0) on Windows
2023/11/18 01:56:16: Plugged in
2023/11/18 01:56:17: Opened serial port
2023/11/18 01:56:18: MSP command failed: Timed out after 1000ms
2023/11/18 01:56:21: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 01:56:22: MSP command failed: Timed out after 1000ms
2023/11/18 01:56:22: Port already in use by another application - try re-connecting
2023/11/18 01:57:51: Opened serial port
2023/11/18 01:57:52: MSP command failed: Timed out after 1000ms
2023/11/18 01:57:54: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 01:57:55: MSP command failed: Timed out after 1000ms
2023/11/18 01:57:55: Port already in use by another application - try re-connecting
2023/11/18 01:58:30: Port selected
2023/11/18 01:58:33: Opened serial port
2023/11/18 01:58:34: MSP command failed: Timed out after 1000ms
2023/11/18 01:58:37: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 01:58:38: MSP command failed: Timed out after 1000ms
2023/11/18 01:58:38: Port already in use by another application - try re-connecting
2023/11/18 02:01:00: Chrome (119.0.0.0) on Windows
2023/11/18 02:01:00: Plugged in
2023/11/18 02:01:24: Opened serial port
2023/11/18 02:01:25: MSP command failed: Timed out after 1000ms
2023/11/18 02:01:28: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 02:01:29: MSP command failed: Timed out after 1000ms
2023/11/18 02:01:29: Port already in use by another application - try re-connecting
2023/11/18 02:01:55: Opened serial port
2023/11/18 02:01:56: MSP command failed: Timed out after 1000ms
2023/11/18 02:01:58: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 02:01:59: MSP command failed: Timed out after 1000ms
2023/11/18 02:01:59: Port already in use by another application - try re-connecting
2023/11/18 02:02:32: Opened serial port
2023/11/18 02:02:33: MSP command failed: Timed out after 1000ms
2023/11/18 02:02:36: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 02:02:37: MSP command failed: Timed out after 1000ms
2023/11/18 02:02:37: Port already in use by another application - try re-connecting
2023/11/18 02:02:41: Opened serial port
2023/11/18 02:02:42: MSP command failed: Timed out after 1000ms
2023/11/18 02:02:44: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2023/11/18 02:02:45: MSP command failed: Timed out after 1000ms
2023/11/18 02:02:45: Port already in use by another application - try re-connecting
2023/11/18 02:03:04: Unplugged
2023/11/18 02:09:10: Chrome (119.0.0.0) on Windows
2023/11/18 02:09:11: Plugged in
2023/11/18 02:09:13: Opened serial port
2023/11/18 02:09:13: MultiWii API version received: 1.45.0
2023/11/18 02:09:13: Flight controller info, identifier: BTFL version: 4.4.3
2023/11/18 02:09:13: Running firmware released on: Nov 14 2023 16:10:10
2023/11/18 02:09:13: Board: S405, version: 0
2023/11/18 02:09:13: Unique device ID received - 0x03f002c3033470f37393338
2023/11/18 02:09:20: Trying to read 0 ESCs

stylesuxx, 2023-11-19:
"Trying to read 0 ESCs" is an indicator that the resources are mapped wrongly on your flight controller. Can you spin motors from within BF configurator? I woould recommend you back up your settings and re-flash betaflight and try with the defaults.

stylesuxx, 2023-11-28:
@shamssh123 please provide more info or I will close this issue.