stylesuxx/esc-configurator #429: Bluejay: support for pauses in melodies

ggoraa, 2023-11-24:
**Is your feature request related to a problem? Please describe.**
I was making a melody for a Bluejay ESC, and I encountered a problem while trying to make a simple "delay" without any sound, but found out there is no way to do so.

**Describe the solution you'd like**
Looking at Nokia's syntax, there is an option for not making sound, example:
```
8b4,2-,4a3
```
Where the 2- signifies a pause, with the number 2 being the delay length
**Describe alternatives you've considered**
Just slapping `e3`s around instead of `-`

**Additional context**
No context

stylesuxx, 2023-11-24:
I think `p` is what you are looking for.

ggoraa, 2023-11-24:
Oh, understood
Yeah, that worked fine,
In that case, we need some actual documentation that explains all of the syntax for melodies :)

stylesuxx, 2023-11-24:
It's really just RTTTL as described here on Wikipedia for example: https://en.wikipedia.org/wiki/Ring_Tone_Text_Transfer_Language

closed 2023-11-24 by stylesuxx