stylesuxx/esc-configurator #435: ESC NOT recognized.

Dingl3berry, 2023-12-14:
i have the SkyStar FC ESC Stack F7 mini HD Pro AM32 bit. when i connect it to the ESC-Configurator its NOT recognized. its a brand new stack just purchased thru pyrodrone. i need to change the motor direction on Motor 2 & 3 but im having an isssue connecting the FC ESC stack . im going to try and attach the error 

[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/13669139/esc-configurator-log.txt)

Dingl3berry, 2023-12-14:
any help would be appreciated Thank You soooo much! if you need any other info please don't hesitate to ask thanks again!

stylesuxx, 2023-12-14:
Battery is plugged in, right? Have you tried reading the ESCs a second time after the first read failed?

stylesuxx, 2023-12-14:
By retry, I mean, you read - it fails, you wait 20 sec and try to read again, while leaving it plugged in. in-between reads. Have you tried the stand alone AM32 configurator from here: https://drive.google.com/file/d/1qw2f7nM6OZftIE5GTgMPWB5zXCELxYG9/view?usp=sharing ?

Dingl3berry, 2023-12-14:
I really do appreciate your help, Thank You! 
yes i can spin the motors in betaflight. and yes i did retry reading the esc after 20,30,40 seconds failure message while still plugged in to power. i tried rebooting my computer and esc stack, I tried a different USB port, I tried a different USB plug, I tried plugging in esc with power first then plugin into computer BUT  I get the same results and that error message i attached above

Dingl3berry, 2023-12-14:
[image]
[image]

Dingl3berry, 2023-12-14:
i tried attaching pics above from the stand alone AM32 configurator but it looks like they didn't come out not sure why.  but basically the configurator will show the 4 boxes numbered 1-4 referring to the motors but when i click on one of the boxes for example motor 2 it reads "M2: Did Not Connected" and nothing happens after that!  oh and at the very bottom status it reads:  "BAD OR NO ACK FROM IF"

stylesuxx, 2023-12-15:
Hmm, that's very interesting. So it's a general problem with your stack rather than with either of the configurators. Have you spoken to Skystar already? Could you link the specific product you bought? Do you have a different computer you could try on (just to exclude that as a factor, I don't really think this will be an issue, just to be on the safe side)?

Dingl3berry, 2023-12-16:
I'm coming to the same conclusion that it's NOT a configurator issue it's a hardware issue! Thank you for your assistance I will follow up with skystars! We close this claim and I thank you again! 🙏🏼

Dingl3berry, 2023-12-16:
https://pyrodrone.com/collections/stacks/products/skystars-f722hd2020-flight-controller-km55-55a-esc

This is the link to the FC stack from PyroDrone.

Dingl3berry, 2023-12-16:
I will try a different computer when I go to my friends house for I only have one computer! And I'll give you an update when I get a chance to try another computer! Thank you!

stylesuxx, 2024-01-11:
Closed due to inactivity.

closed 2024-01-11 by stylesuxx