stylesuxx/esc-configurator #437: Layout Revision 206 Not Yet Supported

Nikonrikam, 2023-12-29:
What is the deal with this error? I have been flashing these ESC's with this site for months now and today I get a failed flash and have no idea why. 

[image]

I am flashing Bluejay, O-H-5, Version 19.2 like I have done before. I use this to swap between 24kHz and 48kHz tunes. I even tried Version 18 of Bluejay and got the same error. 

The FC is flashed to Betaflight 4.4.3.

Thanks,
Sean

Nikonrikam, 2023-12-29:
https://flywoo.net/products/goku-versatile-f411-vtx

This is the AIO FC I am using.

Nikonrikam, 2023-12-29:
[esc-configurator-log (2).txt](https://github.com/stylesuxx/esc-configurator/files/13795836/esc-configurator-log.2.txt)
[esc-configurator-log (1).txt](https://github.com/stylesuxx/esc-configurator/files/13795837/esc-configurator-log.1.txt)
[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/13795838/esc-configurator-log.txt)
My debug logs.

stylesuxx, 2023-12-30:
Flash failed on ESC1. Re-flash using the "ignore MCU layout" checkbox and you should be fine.

stylesuxx, 2024-01-11:
Closed due to inactivity.

closed 2024-01-11 by stylesuxx