stylesuxx/esc-configurator #438: Read ESC 4: UNKNOWN

SamsonsRiddle, 2024-01-03:
Hello and Happy New Years to all. I am here in hopes to find someone to help me get my rig (5" IFlight Nazgul w/ iFlight Succex-e, F405 30x30 FC) running 2207 1800KV) flying, again. I was in the process of testing/comparing and flashing between BlueJay and BlHeli running MultiShot, when it could no longer recognize ESC #4. I've tried using the standalone BlHeli Configurator/App, online configurator as well as flashing to JESC, using its standalone configurator App, with no luck. 

If there is any other info I have left out or you require, please let me know. Ive attached the 'Debug' log and appreciate any suggestions/input/ideas...thank you for your time and any help.

[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/13824404/esc-configurator-log.txt)

stylesuxx, 2024-01-04:
From your log it seems like the ESC is still showing up, but as unknown. You should still be able to flash via configurator, just check the "Ignore MCU layout" checkbox and make sure to select the correct layout yourself.

SamsonsRiddle, 2024-01-04:
Chris, thank you sooo much for your prompt reply and thank you even more for the tip...worked just as you said...thank you and have a wonderful 2024. - cheers

closed 2024-01-04 by stylesuxx