stylesuxx/esc-configurator #439: FLASH writes hanging with a timeout in Betaflight when using Chrome on MacOS

SteveCEvans, 2024-01-06:
Attempting to FLASH ESCs using Chrome on Mac 10.15.7 frequently hangs, bricking the ESC. No issues are seen using Edge on Windows 10.

This has been seen when flashing EFM8BB51 based ESCs necessitating recovery using the C2 interface.

See https://github.com/betaflight/betaflight/pull/13277 which adds timeouts to aid to alleviate this.

stylesuxx, 2024-01-06:
Is this limited to BB51?
When has this first popped up - in regards to MacOS/Chrome version?

I have no Mac myself, so I can't really aid in debugging this one at all.

Could you - by any chance - post a log of a failed flash with the extended debugging setting checked?

SteveCEvans, 2024-01-06:
Note that the same FC with an STM32F405 is unaffected by this issue. It appears to be peculiar to the combination of MacOS and the AT32F435.

I'll grab a log.

SteveCEvans, 2024-01-06:
[esc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/13851266/esc-configurator-log.txt)

SteveCEvans, 2024-01-06:
I first noticed this issue when flashing an AIO with an AT32F435. I've built a bed of nails fixture to be able to recover this esc, but haven't tried on an another ESC as I've yet to wire one up that I can recover if bricked.

stylesuxx, 2024-01-06:
Gotcha - very interesting, thank you.
Have you also tried on some kind of Linux flavor, or just Windows and Mac?

Which AIO are you running?

SteveCEvans, 2024-01-06:
https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-12a-aio-brushless-flight-controller-v3-0

SteveCEvans, 2024-01-06:
I have an Ubuntu 20.04 VM and that seems to work OK with Chrome.

SteveCEvans, 2024-01-06:
Both the Windows 10 and Ubuntu 20.04 are running under Parallels on the Mac which has the issue, so it's not a hardware problem.

SteveCEvans, 2024-01-06:
If we can create a stress test that won’t brick the ESC, that would be good. Would it be possible to erase and write an unused sector or two so a failed write won’t spill our day.