stylesuxx/esc-configurator #441: Esc configurator keeps thinking radio is connected

tjcorrell, 2024-01-08:
trying to flash bluejay to esc and configurator thinks my tx is connected when its not. ive tried everything. no other ports open-> fresh restart computer, first thing i do is try to flas and still thinks tx is connected. im not sure if this is a me problem or a bug but i cant get it flash.
[esc-configurator-log (1).txt](https://github.com/stylesuxx/esc-configurator/files/13856704/esc-configurator-log.1.txt)

stylesuxx, 2024-01-08:
Make sure to set your radio up first. Only after successful Radio configuration we can detect the radio to be off.

tjcorrell, 2024-01-09:
thanks for reply.. tx is setup. everything works in BF. What are possible reasons of 'Layout mismatch, override not enabled - aborted' I still have warning of tx. Does that have anything to do with it?

stylesuxx, 2024-01-09:
Post log and a screenshot please.

stylesuxx, 2024-01-09:
Just checked your log. Your ESCs are recognized as  Q-H-40. Why are you trying to flash G-H-50?

tjcorrell, 2024-01-09:
ah well.. I am a 3d heli guy building his first drone and don't know what im doing. The build i was following was saying to flash gh-50 and 48khz. Is this unnecessasary? The qh40 is 24 khz. should I change this?

stylesuxx, 2024-01-09:
You should keep what is auto detected. Flashing a different layout will - most definitely - brick your ESCs. If you don't know what you are doing yet, why are you trying to flash ESCs? Which problem are you trying to solve? Unless you are flying a 1S whoop, you will most likely not feel a difference on 48kHz.

I don't want to discourage you, but upgrading to the latest and greatest is not always necessary, unless you are looking for a specific feature.

tjcorrell, 2024-01-09:
10-4! thanks for your replies. My apologies for dumb questions lmao

stylesuxx, 2024-01-09:
Don't apologize for asking questions.

Just from experience: If your current setup is working, you are not missing a feature and the software is not limiting you, just have fun flying. Breaking things while flying is much more fun than breaking things through bricking them ;-)

closed 2024-01-11 by stylesuxx