stylesuxx/esc-configurator #443: Can't flash un BLHeliS ESC FVT Littlebee 30A

ChrstianB, 2024-01-11:
Hello, i'm trying to flash Bluejay on my FVT Littlebee 30A ( F395 processor ), but the configurator does not recognize them

[image]

And also i can't flash anything, i can chose only BLHeli , esc model , firmware version but it wont flash, like the flash button does anything

[image]

I did a test with the BLHeliS Configurator, wich works, after finding on rcgroups a guy who flashed A-H-25 target on them.
I tryed to flash it with the BLHelis Conf and it worked (BLHelis saw Bluejay version of the firmware), but after rebooting it seems they got briked ... so i unbricked them with the C2 method (Arduino) and now i'm back on start where i can't flash anything other then BLHeli ( with its configurator).
When unbriking i find out that mine has different processor so it was obiviusly a target problem
[image]
(I did use the BLHeliSuit version 16 as the 32 didn't work...)
Logs
[esc-configurator-log (1).txt](https://github.com/stylesuxx/esc-configurator/files/13899532/esc-configurator-log.1.txt)
[Log blhelis.txt](https://github.com/stylesuxx/esc-configurator/files/13899534/Log.blhelis.txt)

I've tryed with different PC with win 10 and 11 but the problem stays,also tryed with the "ignore esc label option" but nothing change... if we can find out wich target should i use , will it be possible to flash them?

Thanks for help

stylesuxx, 2024-01-11:
AFAIK they are BLHeli ESCs not BLHeli_S. Also one of your screenshot indicates wrong MCU. And the drop-down where you could only chose BLHeli should have been an indicator. Bluejay will only work with EFM8 BB1/BB2/BB51 MCUs.

ChrstianB, 2024-01-11:
Ok so i'll give a try to BLHeli_M then, and if does not works i thinks it's time to change esc ;)
Thanks for helping me figure it out

stylesuxx, 2024-01-11:
Would be surprised if it did, but keep us posted please.

closed 2024-01-11 by stylesuxx