stylesuxx/esc-configurator #451: flash esc

Hasselfpv, 2024-01-22:
When trying to flash the esc with Bluejay with 0.19.2 the following error occurs: Layout revision 206 is not yet supported

stylesuxx, 2024-01-22:
Which ESC?

Hasselfpv, 2024-01-22:
its the iflight Blitz Mini E55S 4-in-1 2-6s ESC , try with macbook and windows. Only esc1 the other went fine.

Hasselfpv, 2024-01-22:
> Which ESC?

its the iflight Blitz Mini E55S 4-in-1 2-6s ESC , try with macbook and windows. Only esc1 the other went fine.

stylesuxx, 2024-01-22:
That's very strange - this is a BLHeli_32 ESC. Please link me to the product page.

stylesuxx, 2024-01-22:
Interesting. If only one failed to flash but is still showing up in the configurator as "Unknown" you can re-flash it. Just check "Ignore MCU layout" checkbox and select the correct layout yourself.

Please post the log if you have any other issues.

Hasselfpv, 2024-01-22:
> Interesting. If only one failed to flash but is still showing up in the configurator as "Unknown" you can re-flash it. Just check "Ignore MCU layout" checkbox and select the correct layout yourself.
> 
> Please post the log if you have any other issues.
Thanks, I'll try it later or tomorrow morning and report back

Hasselfpv, 2024-01-23:
> Interesting. If only one failed to flash but is still showing up in the configurator as "Unknown" you can re-flash it. Just check "Ignore MCU layout" checkbox and select the correct layout yourself.
> 
> Please post the log if you have any other issues.

Now with the check box selected its still work , big thanks for your help.

Hasselfpv, 2024-01-23:
Now i want to enable RPM Filtering , motors are spinning in mo
[image]
tor test , but two of them show error rate

stylesuxx, 2024-01-23:
Hey, that's not really an esc-configurator issue. I suggest you hip on the BF discord and try to get help there. I would just check how the error rate behaves if you spin the motors up (without props of course and not full throttle).

closed 2024-01-23 by stylesuxx