stylesuxx/esc-configurator #457: SpeedyBee F405 V3 stack RPM Reporting 0

RichC75, 2024-02-25:
**Describe the bug**
0 RPM is reporting 
Betaflight 10.10.0-RC2 (norevision)
BluJay updated ESCs
FC Updated to latest ver
Dshot300

When I spin the motors by hand I get an error of .03-.01% which I understand is normal but I do not get an RPM value reported.

When I arm I can hear the motors spin up and stabilize but is there a way to verify they are reporting correctly? Is this normal?

stylesuxx, 2024-02-25:
You get an Error, but no RPM? Makes no sense to me.
Which BF version are you running? Also try with a stable version of BF Configurator to  make sure there is no mismatch...

RichC75, 2024-02-25:
Betaflight 10.10.0-RC2 (norevision)

Had same issue with BF 4.4.3

stylesuxx, 2024-02-25:
One is the version of the  configurator, the other is the version of Betaflight itself. Pleas try with the latest STABLE version of the BF configurator.

Also I think you'd get quicker help in the BF discord since this is not really an issue related to esc-configurator itself...

stylesuxx, 2024-02-25:
OK, that's BF 4.5.0, so that at least should match the Configurator. Please speak to BF support, I can't really help you here, apart from (again) asking you to check with the latest stable releases of both.

closed 2024-02-25 by stylesuxx