stylesuxx/esc-configurator #460: Port already in use by another application - try re-connecting

pjkirby31, 2024-03-17:
I'm trying to connect to the ESCs of my SpeedyBee flight controller, and I've been unable to connect to the ESCs. The SpeedyBee website recommends using this to configure the ESCs, but it won't work for me.

When opening the Port Selection, these are the options coming up, I'm using COM6
[image]

**Log**
```
2024-03-17 @ 16:03:15 -- Port already in use by another application - try re-connecting
2024-03-17 @ 16:03:15 -- Failed resetting ESC 1 - power cycle and try again
2024-03-17 @ 16:03:08 -- Opened serial port
2024-03-17 @ 16:03:07 -- Port selected
2024-03-17 @ 16:00:01 -- Plugged in
2024-03-17 @ 16:00:01 -- Chrome (122.0.0.0) on Windows
```

-   OS: Windows 10
-   Browser: Chrome (Version 122.0.6261.129), Edge (Version 122.0.2365.92)
-   FC: SpeedyBee F405 V3
-   FC Firmware: Arducopter 4.4.4
-   ESC Firmware: BLHeli_S JH50

stylesuxx, 2024-03-18:
Make sure no other program is claiming the serial device, any other configurator or for example cura.

pjkirby31, 2024-03-18:
I looked on device manager and the only thing being used by this port is ardupilot, which is the same port I use to connect to ardupilot with mission planner. Mission planner isn't running so the port shouldn't be used right now.

I also tried connecting on another PC and had the same issue.

[image]

stylesuxx, 2024-03-18:
Could you please verify that you have the same issue with Betaflight flashed on that board.

pjkirby31, 2024-05-23:
After switching to Betaflight, I was able to flash the board. I used a fresh board to do this so maybe that helped too. Thanks for the help!!

closed 2024-05-23 by stylesuxx