stylesuxx/esc-configurator #461: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ version '16.7' (Layout 33) is not yet supported.

bendyna, 2024-04-06:
I've connected SpeedyBee F405 V3 stack + 4 motors (TMotor Velox V2207.5 V2 Veloce Series V1950) to esc-configurator. First I've tried to control all manually via slider, all 4 worked. Then I've pressed "Read setting" and it showed ESC1 as Unknown and all others were correctly identified. I re-connected and after some time ESC1 and ESC3 were unknown. I flashed firmware to all 4 ESCs, 2 were successful and for Unknowns nothing happened (see below log). 
In manual mode they ESC2 and ESC4 now properly work while ESC1 and ESC3 wiggle (only if I put slider to max)

[image]

**Log**
```
2024-04-06 @ 12:24:41 -- Flashed ESC 1 - 0.1s
2024-04-06 @ 12:24:41 -- Layout mismatch, override not enabled - aborted
2024-04-06 @ 12:24:41 -- Flashing ESC 1
```

stylesuxx, 2024-04-06:
If it came like that out of the box, I would have personally returned it. Anyway, check the "ignore MCU layout" checkbox and flash the two unknown ones with the correct firmware.

bendyna, 2024-04-06:
@stylesuxx Thanks a lot, that helped!

For others who can encounter this, checkbox appears after you press "Flash Firmware" on any ESC

closed 2024-04-06 by bendyna

syaeither, 2025-11-04:
what if the checkbox doesn't work? and it's not possible to reflash?