stylesuxx/esc-configurator #462: Port already in use by another application - try re-connecting

dirtybirdnj, 2024-04-23:
Hello I am trying to use this website to configure a brushed ESC I got from just cuz robotics

stylesuxx, 2024-06-12:
Sorry, I somehow missed this issue - could you get it going, or do you still need help?

dirtybirdnj, 2024-06-15:
Thank you for coming back to this! 👏 

I was dead in the water so I've moved on to other stuff but I will do whatever I can to help diagnose / reproduce the issue.

Let me know what you need me to do / info to provide

GrmlMaster, 2024-07-27:
I have the same issue trying to access an AM32 ESC from AliExpress (NEEBRC AM32 80A)

https://de.aliexpress.com/item/1005006877273669.html?spm=a2g0o.productlist.main.17.7ff7MxqAMxqAms&algo_pvid=fb326680-9506-4174-8b69-131d84dace43&algo_exp_id=fb326680-9506-4174-8b69-131d84dace43-8&pdp_npi=4%40dis%21EUR%2120.00%210.93%21%21%21153.41%217.14%21%40211b80d117221032826922331e2aea%2112000038599669257%21sea%21DE%210%21ABX&curPageLogUid=P6tCHrZwLIKt&utparam-url=scene%3Asearch%7Cquery_from%3A

The problem occurs with Edge and Chrome.
The "Multi ESC Config Tool 1.82" can access the ESC.
am32.ca says it can connect but the configuration screen remains empty.

[image]
[image]

dirtybirdnj, 2024-07-27:
Just chiming in to say I am also still stuck

rgreenaus, 2024-08-16:
Adding that this problem also affects me.

I have tried different combinations of: 2 types of ESC, 2 different flight controller models, windows and linux OS, edge and chrome. Same issue on all.

stylesuxx, 2024-08-16:
For AM32 you really should be using [am32.ca](https://am32.ca) - AM32 devs decided to run their own configurator, so no updates will be happening here anymore.

Still, the "Port already in use" error is exactly what it says - there is really nothing on our end that can be done. Make sure no other applications are claiming the device, no other application is open, double check USB cables/try different ones, try different USB ports.

avenstewart, 2025-08-13:
@stylesuxx I'm hitting the same issue with a BLHeli_S 55A 4-in-1 ESC from Speedybee. Tried on three different machines using ubuntu, windows, and mac, using two different USB cables and both USB2.0 and 3.0 ports on two of those machines. Tried it with battery power being supplied and without.

Gbrey925, 2026-02-12:
Just found  a solution for this, for people who are using Ardupilot. You need to enable BLHeli32 Passthrough in Ardupilot for it to work. All of the steps are listed in the documentation: https://ardupilot.org/copter/docs/common-blheli32-passthru.html.

Hope that helps :))