stylesuxx/esc-configurator #464: モーターのdirectionをreverse できなくしてしまいました。

ichyaro, 2024-05-04:
お世話になります。機体のMODELが,HAPPY MODELのMOBULA8 ELRS2.4GHZです。
モーターのdirectionが、normalだったので、reverseに成功していましたが、normaiに戻ってしまいました。
ESC のターゲットがZ-H-30.です。.ESC-1,2,3のVersionを0.19.２へフラッシュできません。
ESC-1　 UNKOWN
ESC-2 　UNKOWN
ESC-3 　UNKPWN
ESC-4　Z-H-30 - ブルージェイ、0.19、48kHz
となります。よく理解しないで、フラッシュしたのが、良くなかったです。

よろしくご指導お願いいたします。

sc-configurator-log.txt](https://github.com/stylesuxx/esc-configurator/files/15209516/esc-configurator-log.txt)

stylesuxx, 2024-05-04:
English please. It seems. something during flashing went wrong. Make sure your batteries are fully charged and flash again, checking the "Ignore MCU Layout" checkbox - make sure to select the correct layout manually.

ichyaro, 2024-05-04:
You want to reverse the direction of the motor, and you were able to successfully reverse it with BLHELI-S. 2024年5月4日 20:59、Chris L. ***@***.***>からのメール： English please. It seems. something during flashing went wrong. Make sure your batteries are fully charged and flash again, checking the "Ignore MCU Layout" checkbox - make sure to select the correct layout manually.

ichyaro, 2024-05-04:
I apologize for not speaking English earlier. Please excuse my poor English.

stylesuxx, 2024-05-05:
No worries. As mentioned before - you need to reflash, selecting the target manually. and checking the "Ignore MCU layout" checkbox. Make sure your batteries are fully charged and probably use a different USB cable than the one that lead to the failed flash.

ichyaro, 2024-05-06:
Thank you very much for your guidance. I didn't do what I was told and caused damage to the FC due to my own selfish actions. There are two reasons for the failure. One is compatible with 1S-2S Lipo/LIHV batteries, but I accidentally connected a 3S Lipo/LIHV battery. My computer can no longer be turned on. I took it to a shop right away, had it diagnosed, removed the battery, and it started up. Second, the esc specifications are BLHeliSuite programmable. Support firmware Target: Z_H_30 Default protocol: DSHOT300 Bluejay firmware It was supported. But be warned: the code memory will be erased. However, in BLHeliSuite, do you want to flash target esc#1 unknown esc (unknown layout) to :z-H-30(MULTI)BLHELI REV:16.7? I checked "Yes" . This is the number one factor that causes undesirable results. In light of this failure, if you are not confident, do not flash ESC, ask others for Heip and wait for advice. However, thanks to the checkbox "Do you want to ignore inappropriate MCUs and layouts?", I was able to find out. This was a valuable achievement, so I am very grateful. I will continue to work hard on the ESC program so that I can fly the drone without giving up. 2024年5月6日 0:46、Chris L. ***@***.***>からのメール： No worries. As mentioned before - you need to reflash, selecting the target manually. and checking the "Ignore MCU layout" checkbox. Make sure your batteries are fully charged and probably use a different USB cable than the one that lead to the failed flash.

closed 2024-05-17 by stylesuxx