stylesuxx/esc-configurator #466: Issue connecting my Speedybee esc to esc configurator

cooperthu, 2024-05-18:
Hi, im currently in the process of building my first fpv drone.  I am currently trying to configure the esc but esc configurator keeps telling me that the port is already in use for another application.  I don't have betafpv or anything else running in the backround and was wondering what I shoudl do to troubleshoot.

stylesuxx, 2024-05-18:
Some other application is most likely claiming the serial port. Cura for example likes to do that. I assume you have no issue connecting to Betaflight Configurator, right?

cooperthu, 2024-05-18:
I hadn't tried to connect it to betaflight, but it appears that doesn't work either.
Could it be an issue with the usb port on the esc?

stylesuxx, 2024-05-18:
The ESC itself should not have a USB port. Which hardware are we talking about? Please link. Also which operating system are you running? Are you sure all required drivers are installed? You have tried different USB cables?

cooperthu, 2024-05-18:
Im using the Speedybee v3 flight controller and esc stack.
[image]

I tried a few different cables and none of them work

stylesuxx, 2024-05-18:
You did not answer the rest of my questions. If it's not a driver issue, and no device shows up in the device manager your stack might have a general problem... But check your device manager and drivers first

stylesuxx, 2024-06-12:
@cooperthu - any update on this, otherwise I will close this issue.

stylesuxx, 2024-07-19:
Closing due to inactivity.

closed 2024-07-19 by stylesuxx