stylesuxx/esc-configurator #469: USB does not connect

lotaloi, 2024-06-19:
HI developers, 
Sometimes the configurator does not get connected with the USB , saying a wrong message " com port is used by another application" . 
Pls would u fix that and thanks.

stylesuxx, 2024-06-20:
There is nothing really we can fix on our end. Sometimes other applications can claim the serial port, if that's the case, the browser can no longer claim it.

The only remedy is to close all apps that might claim the serial port, Cura for example is known to do that.

If you are suspecting that the message is a false positive, please post the log, also post output from `chrome://device-log`. (Clear before attaching device, refresh after error occurs).

stylesuxx, 2024-07-09:
Closing due to inactivity.

closed 2024-07-09 by stylesuxx