stylesuxx/esc-configurator #478: IFlight BLITZ F435 uptade to Bluejay fail

A-ZuMa, 2024-08-01:
I tried to update the BLITZ F435 firmware to Bluejay 19.1, but after starting the process, everything stopped at about 20%. After rebooting, the configurator does not see the failed motor. I started update firmware again, the second channel was flashed quickly, but on the third is fail, it stopped and the firmware is broken.
The configurator no longer sees 2 channels. :(
I installed BlHeli configurator and it saw broken 2 channels, which I was able to flash on BlHell firmware.
I connected the configurator again through ESC configurator and tried to flash Bluejay, but with every attempt, the firmware stops after starting. :(
[image]

stylesuxx, 2024-08-01:
Please post the log from esc-configurator.com

A-ZuMa, 2024-08-01:
> Please post the log from esc-configurator.com

2024/08/01 17:54:40: Plugged in
2024/08/01 17:54:42: Opened serial port
2024/08/01 17:54:42: MultiWii API version received: 1.46.0
2024/08/01 17:54:42: Flight controller info, identifier: BTFL version: 4.5.0
2024/08/01 17:54:42: Running firmware released on: Aug 10 2023 07:39:06
2024/08/01 17:54:42: Board: A435, version: 0
2024/08/01 17:54:42: Unique device ID received - 0x048a8375a30000d790711
2024/08/01 17:54:46: Trying to read 4 ESCs
2024/08/01 17:54:46: Layout revision 208 is not yet supported
2024/08/01 17:54:46: Read ESC 1: UNKNOWN
2024/08/01 17:54:49: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/08/01 17:54:49: MessageNotOkError: Message not OK 
2024/08/01 17:54:49: Failed reading ESC 2
2024/08/01 17:54:49: Read ESC 3: Q-H-25 - Bluejay, 0.19, 48kHz
2024/08/01 17:54:50: Read ESC 4: Q-H-25 - Bluejay, 0.19, 48kHz
2024/08/01 17:54:50: Done reading ESCs
2024/08/01 17:55:12: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/Q_H_25_48_v0.21.0.hex 
2024/08/01 17:55:12: Layout revision 208 is not yet supported
2024/08/01 17:55:12: Flashing ESC 1
2024/08/01 17:55:12: Layout mismatch, override not enabled - aborted
2024/08/01 17:55:12: Flashed ESC 1 - 0.1s
2024/08/01 17:55:13: Layout revision 208 is not yet supported
2024/08/01 17:55:33: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/Q_H_25_48_v0.21.0.hex 
2024/08/01 17:55:34: Flashing ESC 3
2024/08/01 17:55:34: ### Step 1: Erasing EEPROM and writing safeguards 
2024/08/01 17:55:34: ### Step 2: Writing bootloader failsafe (brick protection) 
2024/08/01 17:55:34: ### Step 3: Erase, write and verify Pages 0x02-0x0D 
2024/08/01 17:55:42: ### Step 4: Erase, write and verify Pages 0x00-0x02 
2024/08/01 17:55:43: ### Step 5: Write EEPROM section 
2024/08/01 17:55:44: Flashed ESC 3 - 10.5s
2024/08/01 17:55:44: Bluejay layout found 
2024/08/01 17:55:44: Migrated setting RPM_POWER_SLOPE 
2024/08/01 17:55:44: Migrated setting MOTOR_DIRECTION 
2024/08/01 17:55:44: Migrated setting BRAKING_STRENGTH 
2024/08/01 17:55:44: Migrated setting COMMUTATION_TIMING 
2024/08/01 17:55:45: Migrated setting BEEP_STRENGTH 
2024/08/01 17:55:45: Migrated setting BEACON_STRENGTH 
2024/08/01 17:55:45: Migrated setting BEACON_DELAY 
2024/08/01 17:55:45: Migrated setting DEMAG_COMPENSATION 
2024/08/01 17:55:45: Migrated setting BRAKE_ON_STOP 
2024/08/01 17:55:45: Migrated setting LED_CONTROL 
2024/08/01 17:55:45: Migrated setting POWER_RATING 
2024/08/01 17:55:45: Migrated setting STARTUP_MELODY 
2024/08/01 17:55:45: Updating ESC 3 - finished
2024/08/01 17:55:49: Trying to read 4 ESCs
2024/08/01 17:55:49: Layout revision 208 is not yet supported
2024/08/01 17:55:49: Read ESC 1: UNKNOWN
2024/08/01 17:55:50: Read ESC 2: **FLASH*FAILED**
2024/08/01 17:55:51: Read ESC 3: Q-H-25 - Bluejay, 0.21.0, 48kHz
2024/08/01 17:55:51: Read ESC 4: Q-H-25 - Bluejay, 0.19, 48kHz
2024/08/01 17:55:51: Done reading ESCs
2024/08/01 17:56:03: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/Q_H_25_48_v0.21.0.hex 
2024/08/01 17:56:04: Flashing ESC 2
2024/08/01 17:56:04: ### Step 1: Erasing EEPROM and writing safeguards 
2024/08/01 17:56:04: ### Step 2: Writing bootloader failsafe (brick protection) 
2024/08/01 17:56:04: ### Step 3: Erase, write and verify Pages 0x02-0x0D 
2024/08/01 17:56:12: ### Step 4: Erase, write and verify Pages 0x00-0x02 
2024/08/01 17:56:13: ### Step 5: Write EEPROM section 
2024/08/01 17:56:14: Flashed ESC 2 - 10.5s
2024/08/01 17:56:20: Trying to read 4 ESCs
2024/08/01 17:56:20: Layout revision 208 is not yet supported
2024/08/01 17:56:20: Read ESC 1: UNKNOWN
2024/08/01 17:56:20: Read ESC 2: Q-H-25 - Bluejay, 0.21.0, 48kHz
2024/08/01 17:56:20: Read ESC 3: Q-H-25 - Bluejay, 0.21.0, 48kHz
2024/08/01 17:56:21: Read ESC 4: Q-H-25 - Bluejay, 0.19, 48kHz
2024/08/01 17:56:21: Done reading ESCs
2024/08/01 17:56:32: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/Q_H_25_48_v0.21.0.hex 
2024/08/01 17:56:32: Flashing ESC 4
2024/08/01 17:56:32: ### Step 1: Erasing EEPROM and writing safeguards 
2024/08/01 17:56:33: ### Step 2: Writing bootloader failsafe (brick protection) 
2024/08/01 17:56:33: ### Step 3: Erase, write and verify Pages 0x02-0x0D 
2024/08/01 17:56:35: Command cmd_DeviceWrite failed: Timed out after 1000ms 
2024/08/01 17:56:36: Command cmd_DeviceWrite failed: Timed out after 1000ms 
2024/08/01 17:56:37: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:38: Command cmd_DeviceWrite failed: Timed out after 1000ms 
2024/08/01 17:56:39: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:40: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:41: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:42: Command cmd_DeviceWrite failed: Timed out after 1000ms 
2024/08/01 17:56:43: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:44: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:45: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:46: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms 
2024/08/01 17:56:47: Command cmd_InterfaceTestAlive failed: Timed out after 1000ms

stylesuxx, 2024-08-01:
Are you sure your USB cable is fine?
Battery is plugged in an fully charged, right?
From what I can see, you managed to flash all apart from the 1st ESC.

I'd try a different USB cable and if you can, a different machine.

stylesuxx, 2024-08-12:
Closing due to lack of information.

closed 2024-08-12 by stylesuxx