stylesuxx/esc-configurator #480: Donate a song:  Fontaine-Genshin Impact

RainityZhou, 2024-08-06:
ESC1：
Melody:b=100,o=5,d=8:4d#6,d#6,4c#6,b,4a#,4a#,d#,f,f#,g#,a#,g#,f#,d#,4f#,f#,4f,f,p,4d#6,4c#6,b,4a#,4a#,d#,f,f#,g#,a#,g#,f#,f,d#

ESC2：
Melody:b=100,o=4,d=8:
d#,16d#5,16p,d#5,c#,16c#5,16p,c#5,d#,16d#5,16p,d#5,d#,16d#5,16p,d#5,
b,16b5,16p,b5,b,16b5,16p,b5,g#,16g#5,16p,g#5,a#,16a#5,16p,a#5,
p,d#,16p,16d#,c#,16c#5,16p,c#5,d#,16d#5,16p,d#5,d#,16d#5,16p,d#5,
b,16b5,16p,b5,c#,16c#5,16p,c#5,d#5

ESC3：
Melody:b=100,o=4,d=8:
p,16a#,16p,a#,p,16g#,16p,g#,p,16a#,16p,a#,p,16a#,16p,a#,
p,16f#,16p,f#,p,16f#,16p,f#,p,16d#,16p,d#,p,16f,16p,f,
4p,p,4c#5,d5,4d#5,d#5,4a#5,a#5,
4g#5,g#5,4c#5,c#5

ESC4：
Melody:b=100,o=4,d=8:
4a#5,a#5,4g#5,f#5,2a#5,f#5,g#5,
2f#5,4f#5,4g#5,g#5,4a#5,a#5,
p,a#,16p,16a#,p,16g#,16p,g#,p,16a#,16p,a#,p,16a#,16p,a#,
p,16f#,16p,f#,p,16g#,16p,g#,a#

RainityZhou, 2024-08-06:
Really Sorry I'm not good at using github, looks like I sent my donation at a wrong place.

RainityZhou, 2024-08-06:
How can I delete this issue?

stylesuxx, 2024-08-06:
No worries, closing this one. Thank you!

closed 2024-08-06 by stylesuxx