stylesuxx/esc-configurator #484: ESC Power Rating setting could have better UX

Maxoper, 2024-08-26:
**Is your feature request related to a problem? Please describe.**
This might be picky, but the ESC Power Rating setting could have better UX. I initially set my SuperX HD ESC to 1S, thinking it was correct since I'm using 1S packs. I experienced occasional startup power issues with motors not spinning up. After a long time, I read the tooltip, changed the setting to 2S, and immediately resolved the problem.

**Describe the solution you'd like**
When selecting 1S it could show some kind of small red text stating something like "1-2S ESC should configure 2S+".

**Describe alternatives you've considered**
The dropdown could also show those 3 options with the 2nd and 3rd option both ultimately mapped to the same configuration in the background:
"1S",  "1-2S",  "2S+"

[image]

Very thankful though and this is definitely no priority stuff!

stylesuxx, 2024-08-26:
I think this setting is tricky without reading the info bubble. But you are right, maybe a third option would make it more intuitive - I am just not a big fan of options that are not really options...

Thank you for your feedback in any case.

Maxoper, 2024-08-26:
> I think this setting is tricky without reading the info bubble. But you are right, maybe a third option would make it more intuitive - I am just not a big fan of options that are not really options...

I think my first "solution" would be indeed cleaner, to just show a small warning when selecting 1S. Even if it is just saying: "check the tooltip/hint".

pfeerick, 2024-08-29:
Perhaps?
- 1S Only
- 1-2S+

Since 2S+ is simply wrong if it encapsulates 1S 🤭

stylesuxx, 2024-08-29:
2S+ never encapsulates pure 1S boards. But I get your point.

So far I like your suggested options most.

stylesuxx, 2024-08-29:
Merged into develop will be part of the next release, thanks again for feedback.

closed 2024-08-29 by stylesuxx