stylesuxx/esc-configurator #485: Motors not spinnig wenn i Arm

jericho125, 2024-08-26:
my motors spinning start spinning smoothly on 1100 on the Motortest tap.
wenn i arm the quad with the radio all four Motors won't spin smoothly is like 1050 on the Motortest tap.

my setting: 48khz 0.19.2 damage low
minimum start up power 1100
start up protection 1200

the esc f405 flywhoo 1-2s air v2
on 1202,5 11500 kv motors

stylesuxx, 2024-08-26:
Please check the setup guide: https://github.com/bird-sanctuary/bluejay/wiki/Setup

Are you running in 1S? What's the dead-time of the ESC?

jericho125, 2024-08-26:
yess iam running 1S were can i Change the deadtime?

stylesuxx, 2024-08-26:
You don't change the dead-time. I just want to know how high it is. When you connect your ESC to the configurator  it will show something like Z-H-30 - the last part is the dead-time. With such big motors on 1S I can imagine you might need to crank startup power and idle up - just follow the setup guide and you should be fine.

jericho125, 2024-08-26:
yes is on 30 Startup is on 1100/1200
idle is on 66 birektional on

stylesuxx, 2024-08-26:
I assume you mean idle is on 6.6 - right? Anyway, as mentioned before, follow the [setup guide](https://github.com/bird-sanctuary/bluejay/wiki/Setup) - higher idle might be necessary.

stylesuxx, 2024-08-26:
Closing this issue here since it's not related to the configurator, if you need further assistance, feel free to join the [Bluejay discord](https://discord.gg/ddyzguPB5t).

closed 2024-08-26 by stylesuxx