stylesuxx/esc-configurator #486: Min/max Default values

stylesuxx, 2024-08-29:
A user on discord reported that restoring defaults sets min Startup Power above max startup power.

closed 2024-08-29 by stylesuxx

TGS78, 2025-04-16:
on my Z-H-30 with 0.21.0 or 0.21.1 RC1 the default for min. startup power is still above max. start power
min default is 1025
max default is 1020

stylesuxx, 2025-04-16:
Even if you reset to default values?

TGS78, 2025-04-16:
yes, when press "Restore Default Settings" in ESC Web Configurator it always goes back to 1025/1020.
I can also check with some other ESCs if you want.

stylesuxx, 2025-04-16:
Ah, wait a minute, this is currently not released. It should work on the develop version though: https://develop.esc-configurator.com/

Could you confirm?

stylesuxx, 2025-04-16:
No worries - good reminder to release a new version. There are a couple other things I want to get done first though...