stylesuxx/esc-configurator #491: Flash failed on HobbyWing BLHeli_S S7X2 XRotor-65A-4in1-Lite-AT-RTF

dyarkovoy, 2024-09-24:
When trying to flash this ESC I get "FourWay.js:1286 Uncaught (in promise) Error: failed to verify write **FLASH*FAILED**" in the browser console.
This happens when trying to flash BlueJay or BLHeli_S. There are no errors in the log, just in the browser console. The app just freezes after the error.

**Log**
```
2024-09-24 @ 18:42:07 -- Flashing ESC 1
2024-09-24 @ 18:38:51 -- Done reading ESCs
2024-09-24 @ 18:38:51 -- Read ESC 4: D-X-30 - BLHeli_S, 16.7
2024-09-24 @ 18:38:50 -- Read ESC 3: D-X-30 - BLHeli_S, 16.7
2024-09-24 @ 18:38:48 -- Read ESC 2: D-X-30 - BLHeli_S, 16.7
2024-09-24 @ 18:38:46 -- Read ESC 1: D-X-30 - BLHeli_S, 16.7
2024-09-24 @ 18:38:45 -- Trying to read 4 ESCs
2024-09-24 @ 18:38:41 -- Unique device ID received - 0x02900273433510b32353235
2024-09-24 @ 18:38:41 -- Board: S7X2, version: 0
2024-09-24 @ 18:38:41 -- Running firmware released on: Nov 28 2022 07:30:19
2024-09-24 @ 18:38:41 -- Flight controller info, identifier: BTFL version: 4.3.2
2024-09-24 @ 18:38:41 -- MultiWii API version received: 1.44.0
2024-09-24 @ 18:38:41 -- Opened serial port
2024-09-24 @ 18:38:40 -- Plugged in
2024-09-24 @ 18:38:40 -- Chrome (129.0.0.0) on Windows

-   OS: Windows
-   Browser Chrome
-   Version 129.0.6668.58 (Official Build) (64-bit)

Browser console: 
[esc-configurator.com-1727192580582.log](https://github.com/user-attachments/files/17117162/esc-configurator.com-1727192580582.log)

dyarkovoy, 2024-09-24:
Here is the log: [esc-configurator-log (1).txt](https://github.com/user-attachments/files/17117464/esc-configurator-log.1.txt)

stylesuxx, 2024-09-24:
Hmm. Yes, definitely issue with writing, but lock byte does not seem to be set.
If you change settings, do those stick? Like changing motor rotation direction for example?

stylesuxx, 2024-09-24:
Interesting. And the problem persists on all of the ESCs? Can you try flashing one specific (not first one) please?

dyarkovoy, 2024-09-24:
Yes, already tried 2 different ESCs, same story. I can try the remaining two if you wish, but I don't think it will make a difference

stylesuxx, 2024-09-24:
Yeah, probably won't make a difference. At this point it's probably easiest to just recover via C2 Interface.

dyarkovoy, 2024-09-24:
Ok, I will probably just leave it with BLHeli_S. Maybe the manufacturer somehow intentionally locked it to prevent updates. I don't have enough time and/or knowledge to fiddle with C2 at the moment, but maybe sometime in the future I will try to solve this puzzle. Thank you for help.

closed 2024-09-24 by dyarkovoy

stylesuxx, 2024-09-24:
OK, please keep us posted. I will see if I can get hold of that ESC and try to investigate...
C2 recovery is really not too dramatic if you have an arduino and the C2 pads are somewhat accessible. Feel free to join us on Discord if you need more guidance.

dyarkovoy, 2024-09-25:
I contacted the manufacturer, it looks like they use modified BLHeli_S. So it may not be a good idea to flash it with the stock version. Let me see if I can get modified Bluejay and flashing procedure from them.

dyarkovoy, 2024-09-25:
This is what I got from the manufacturer:
```
XRotor-65A-4in1-Lite-AT-RTF
This is a 32-bit program developed by us. It cannot be flashed to BLHeli-S or Bluejay. The MCU is 32-bit, not 8-bit BB21
```
Can it be that ESC configurator detects BLHeli_S in 32 bit MCU?

stylesuxx, 2024-09-25:
WTF? Then they majorly fucked up by not having a proper header in the firmware...

Thanks for the heads up anyway. They probably should **TELL** their customers this fact.

How is it then, that you can properly change settings? Where is their configurator?

Bullshit like this just makes me angry, zero communication and waste of everyone's time...

dyarkovoy, 2024-09-25:
> How is it then, that you can properly change settings? Where is their configurator?
I can change settings from ESC-configurator and from Betaflight. Can it be that ESC-configurator works for changing the settings on 32 bit MCU? If not, then maybe it is not 32 bit after all. It is quite possible that the manufaсturer's response was from the wrong person. It happens in China.

stylesuxx, 2024-09-25:
That all depends on the bootloader and firmware.

The configurator initially reads the first page of memory, this is then used to distinguish which firmware  it is. If a firmware identifies as BLHeli, then the configurator assumes it is.

The configurator is really agnostic of platform, so technically it is not able to distinguish if it's an 8bit or 32bit MCU. This information is all deduced from the first page being read. From there the offsets are being set and the rest of the communication are basically just read/write operations via the bootloader.

dyarkovoy, 2024-09-25:
Okay, thank you for explanation. This makes perfect sense. 
I was able to read the marking on the MCU: BB51 F16I. I believe it stands for SiLabls EFM8BB51, which is 8-bit MCU. So it seems that support guy from HobbyWing was wrong. Also I think I found C2 pads on the ESC. As far as I understand, these should be pins 5 and 6 of the MCU (C2CK/C2D), right? I don't have Arduino, but I'm pretty fluent with ESP32, is there an easy way to  use С2 recovery via ESP32?

stylesuxx, 2024-09-25:
If it's a BB51 chip, then it definitely is 8 bit and the guy does not know what he is talking about. Good, that relaxes me a bit ;-)

I don't think there is an easy way with esp to be honest. But it's really pretty straight forward with Arduino if you have an Uno or nano: https://stylesuxx.github.io/arduino-c2-flasher/

dyarkovoy, 2024-09-25:
Understood, I will have to order an Arduino then.
Btw, I tried to flash using BLHeliSuite and got the error - Flash verify failed, so apparently this is not ESC configurator issue and has to do with some sort of write protection by the manufacturer.
I was also able to read the flash contents and compared it with the stock BLHeli 16.7, there are differences, so it is indeed a modified firmware.

stylesuxx, 2024-09-25:
There will be differences depending on how you read it and which settings have been changed. So just diffing hex files might not yield the whole truth.

dyarkovoy, 2024-09-25:
Makes sense. I don't know if that tells you anything, but here are the differences:
[image]

richard087, 2026-09-05:
I have bad news, and worse news.

1. the reporter from https://github.com/stylesuxx/esc-configurator/issues/491#issuecomment-2373732103 is not mistaken. I have the same ESC and the same problem. This ESC uses 4x Artery AT32F421K8U7 MCUs, and definitely does not support Bluejay, and is definitely 32bit.
2. I have not yet figured out how to reverse the \*\*FLASH*FAILED\*\* mess; although my ESC seems to function (?)
3. I cannot update to AM32 firmware (which _should_ work) instead of the proprietary mess that HobbyWing has foisted upon us.