stylesuxx/esc-configurator #492: PROBLEM WITH CONECTION

NovatosDelFpv, 2024-09-25:
[esc-configurator-log.txt](https://github.com/user-attachments/files/17137003/esc-configurator-log.txt)

stylesuxx, 2024-09-25:
Probably BLHelis32? Please provide proper info or I'll close this issue.

NovatosDelFpv, 2024-09-25:
If blhelis32 is correct, how could I pass it to bluejay?

stylesuxx, 2024-09-25:
You can't. Please tell me which ESC you have exactly (link to product page).

NovatosDelFpv, 2024-09-25:
FOXXER REAPER F4 128k 65A BL32

NovatosDelFpv, 2024-09-25:
https://www.fpv24.com/es/foxeer/foxeer-reaper-f4-128k-65a-bl32-4in1-fpv-esc

stylesuxx, 2024-09-25:
Yes that's BLHeli_32. We don't support that. Also it's generally no longer maintained, you can't update that anyway.
See if you can flash AM32, other than that you can use it as it is. Good luck!

closed 2024-09-25 by stylesuxx