stylesuxx/esc-configurator #493: Can ESC-Configurator.com be used with an ESC connected via a USB linker? (I do not own a flight controller)

oeloo, 2024-09-28:
Hi, I am an RC boat fan, so I do not own any flight controller. I need BlueJay to solve a motor shuttering issue and need its music editor feature as it is for my kid

I have successfully updated my [ESC](https://www.aliexpress.com/item/4001222916973.html) BLHeli_S version to the latest BLHeli_S revision (v16.7) last year:
 - by connecting my ESC with my [USB linker](https://www.aliexpress.com/item/1005006054341976.html) to my PC
 [image]
[image]
[image]

 - and using the software 'bitdump BLHeli Suite 16'
[image]

Now I want to replace BLHeli_S on my ESC LittleBee with BlueJay. The recommended method is to flash BlueJay is to use ESC-Configurator.com. So I have plugged my ESC in the same way to my computer and used ESC-Configurator, but I have this error message "Port already in use by another application - try re-connecting"
[image]

Can ESC-Configurator.com be used with an ESC connected via a USB linker? (I do not own a flight controller)
If it is impossible to use ESC-Configurator.com, which tool is recommended for flashing BlueJay over BLHeli_S for people not owning a flight controller? they are many rc car fans and RC boat fans that do not have a flight controller.

stylesuxx, 2024-09-28:
No it can not. Please also keep in mind, that with this particular ESC you are showing we have had users report issues with Bluejay (or rather with DSHOT in general) - your mileage may vary here.

closed 2024-09-28 by stylesuxx

oeloo, 2024-09-29:
Thank you for the information about ESC Configuration and the precious information that people have reported issues with BlueJay for my ESC mode.
My best solution is directly purchasing an ESC with BlueJay on it. Where can I buy an ESC directly with BlueJay? I have looked at AliExpress but a ESC BlueJay search provides no result ! Is it forbidden to sell a ESC with BlueJay on it due to its license? How can I then get a ESC with BlueJay?

stylesuxx, 2024-09-29:
Single ESCs are not really popular in the FPV hobby, so most of the stuff that comes with Bluejay are either AIOs or 4in1 platforms.

If you really need to flash bluejay then I would recommend simply buying some cheapo Flight controller. You will need it anyway if you want to change the startup melody.

Please also be aware that Bluejay only supports DSHOT. As far as I know boats will usually just use PWM as throttle signal and you probably also would like to set the ESC up in such a way that it can go forwards and backwards, which you would need to set up via the configurator anyway.

oeloo, 2024-09-29:
Thank you for pointing out that BlueJay only supports DShot. My boat uses PWM, so I cannot use BlueJay. But then, which up-to-date firmware is there for my ESC with BLHeli_S ? BLHeli_S is no longer maintained

stylesuxx, 2024-09-29:
You can go with Am32 too - those are 32 BIT MCUs. You'd also be able to have melodies and PWM is supported (as far as I remember).