stylesuxx/esc-configurator #494: Speedybee F405 V4 Issue Bluejay

flyneed, 2024-09-29:
Hello,

two stacks, the same issue.
After flashing with bluejay (no problems with flashing, even the melodie editor works) the FC (betaflight 4.51) could not find the ESCs with activated Dshot300 and bididshot.
Maybe you have a solution for this problem.
 
Best regards from Germany
flyneed

stylesuxx, 2024-09-29:
What do you mean by "could not find the ESC"? If BF configurator complains about no ESCs being present, usually (re)flashing the correct target should fix it.

flyneed, 2024-09-30:
Hello,
thanks a lot for your message!
Today I have flashed the FC to to newest BF version again an I do not know why, but like a miracle it worked. In the moment we made our first flight - nearly fully satisfied with our first drone build.
Best regards from Germany
flyneed