stylesuxx/esc-configurator #497: Immersionrc EZESC 16 AMP ESC Configurator Issue

MAAZTAHAA, 2024-11-08:
HI,
   I m facing issue regarding connecting Immersionrc EZESC 16 AMP to ESC configurator. I m nor able to connect it with AM32 configurator neither with universal ESC configurator. FC is Piko Blx and ESC configurator failed to read Esc data.Log file is attached.

2024/11/06 00:01:01: Chrome (130.0.0.0) on Windows
2024/11/06 00:01:48: NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user. 
2024/11/07 18:44:36: Chrome (130.0.0.0) on Windows
2024/11/07 18:45:43: Port selected
2024/11/07 18:45:52: Opened serial port
2024/11/07 18:45:52: MultiWii API version received: 1.35.0
2024/11/07 18:45:52: Flight controller info, identifier: CLFL version: 2.0.5
2024/11/07 18:45:52: Running firmware released on: Jul  9 2017 14:38:52
2024/11/07 18:45:52: Board: PIKO, version: 0
2024/11/07 18:45:52: Unique device ID received - 0x041002d4234570b20363434
2024/11/07 18:47:33: Trying to read 4 ESCs
2024/11/07 18:47:36: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:47:36: MessageNotOkError: Message not OK 
2024/11/07 18:47:36: Failed reading ESC 1
2024/11/07 18:47:39: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:47:39: MessageNotOkError: Message not OK 
2024/11/07 18:47:39: Failed reading ESC 2
2024/11/07 18:47:42: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:47:42: MessageNotOkError: Message not OK 
2024/11/07 18:47:42: Failed reading ESC 3
2024/11/07 18:47:45: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:47:45: MessageNotOkError: Message not OK 
2024/11/07 18:47:45: Failed reading ESC 4
2024/11/07 18:49:18: Trying to read 4 ESCs
2024/11/07 18:49:21: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:21: MessageNotOkError: Message not OK 
2024/11/07 18:49:21: Failed reading ESC 1
2024/11/07 18:49:24: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:24: MessageNotOkError: Message not OK 
2024/11/07 18:49:24: Failed reading ESC 2
2024/11/07 18:49:27: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:27: MessageNotOkError: Message not OK 
2024/11/07 18:49:27: Failed reading ESC 3
2024/11/07 18:49:29: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:30: MessageNotOkError: Message not OK 
2024/11/07 18:49:30: Failed reading ESC 4
2024/11/07 18:49:34: Trying to read 4 ESCs
2024/11/07 18:49:37: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:37: MessageNotOkError: Message not OK 
2024/11/07 18:49:37: Failed reading ESC 1
2024/11/07 18:49:40: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:40: MessageNotOkError: Message not OK 
2024/11/07 18:49:40: Failed reading ESC 2
2024/11/07 18:49:42: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:42: MessageNotOkError: Message not OK 
2024/11/07 18:49:42: Failed reading ESC 3
2024/11/07 18:49:45: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 18:49:45: MessageNotOkError: Message not OK 
2024/11/07 18:49:45: Failed reading ESC 4
2024/11/07 18:49:58: Closed port
2024/11/07 19:32:07: Chrome (130.0.0.0) on Windows
2024/11/07 19:32:07: Plugged in
2024/11/07 19:32:12: Unplugged
2024/11/07 19:32:19: Plugged in
2024/11/07 19:32:30: NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user. 
2024/11/07 19:32:32: Opened serial port
2024/11/07 19:32:32: MultiWii API version received: 1.40.0
2024/11/07 19:32:32: Flight controller info, identifier: CLFL version: 2.5.0
2024/11/07 19:32:32: Running firmware released on: Sep 30 2018 20:09:12
2024/11/07 19:32:32: Board: PIKO, version: 0
2024/11/07 19:32:32: Unique device ID received - 0x041002d4234570b20363434
2024/11/07 19:33:05: Trying to read 4 ESCs
2024/11/07 19:33:08: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 19:33:08: MessageNotOkError: Message not OK 
2024/11/07 19:33:08: Failed reading ESC 1
2024/11/07 19:33:11: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/07 19:33:11: MessageNotOkError: Message not OK 
[... 80 lines cut ...]
2024/11/08 18:08:22: Opened serial port
2024/11/08 18:08:22: MultiWii API version received: 1.35.0
2024/11/08 18:08:22: Flight controller info, identifier: CLFL version: 2.0.1
2024/11/08 18:08:22: Running firmware released on: Apr 18 2017 22:27:05
2024/11/08 18:08:22: Board: PIKO, version: 0
2024/11/08 18:08:22: Unique device ID received - 0x041002d4234570b20363434
2024/11/08 18:09:33: Trying to read 6 ESCs
2024/11/08 18:09:36: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/08 18:09:36: MessageNotOkError: Message not OK 
2024/11/08 18:09:36: Failed reading ESC 1
2024/11/08 18:09:39: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/08 18:09:39: MessageNotOkError: Message not OK 
2024/11/08 18:09:39: Failed reading ESC 2
2024/11/08 18:09:42: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/08 18:09:42: MessageNotOkError: Message not OK 
2024/11/08 18:09:42: Failed reading ESC 3
2024/11/08 18:09:45: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/08 18:09:45: MessageNotOkError: Message not OK 
2024/11/08 18:09:45: Failed reading ESC 4
2024/11/08 18:09:48: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/08 18:09:48: MessageNotOkError: Message not OK 
2024/11/08 18:09:48: Failed reading ESC 5
2024/11/08 18:09:51: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/11/08 18:09:51: MessageNotOkError: Message not OK 
2024/11/08 18:09:51: Failed reading ESC 6

stylesuxx, 2024-11-09:
Is the board powered?

stylesuxx, 2025-01-06:
Any update on this? Otherwise I will close the issue.

stylesuxx, 2025-01-16:
Closing due to inactivity

closed 2025-01-16 by stylesuxx