stylesuxx/esc-configurator #498: Sequre 12200 AM32 No connection

jimmy6616, 2024-11-09:
From the web link, com5 is appearing in the connection options, constantly getting , '2024-11-09 @ 11:48:39 -- Port already in use by another application - try re-connecting' , re-started the laptop, Win11, same issue. No other usb devices attached

stylesuxx, 2024-11-09:
AM32 have their own configurator, please use that https://am32.ca

closed 2024-11-09 by stylesuxx

jimmy6616, 2024-11-09:
Tried that as well , it’s says connected to device , when you click on Configurator and select base it tune nothing happens 
Jim

stylesuxx, 2024-11-09:
The "port already in use" message usually means exactly that: some other application is claiming the serial device. Cura for example likes to do that. If in doubt, I would just try on a different computer.