stylesuxx/esc-configurator #499: Upgrade esc BLS bluejay to PWM

Soares493, 2024-11-12:
Friend I purchased an ESC BLS Bluejay 100A to install on a fixed wing, however it does not work with PWM. I would like help to make the settings. How to connect ESC directly to PC without controller?

stylesuxx, 2024-11-12:
PWM has been deliberately removed from Bluejay. If you need PWM, please run Blheli_S.

closed 2024-11-12 by stylesuxx

Soares493, 2024-11-13:
Thank you for your attention. 
Do you have any tutorial on how to update the firmware from ESC to BlHeli? How to do this without a controller card, i.e. connect the ESC directly to the PC?

stylesuxx, 2024-11-13:
You will need a flight controller with BF passthrough, the ln you can use https://esc-configurator.com

Soares493, 2024-11-18:
Friend, can you help how to upgrade to Blheli_S ?