stylesuxx/esc-configurator #500: Esc configurator error

Wasee045, 2024-11-13:
The amount of available ESCs does not match what the flight controller is reporting. This can have multiple reasons:
The ESC was not ready yet. This can for example happen if it is playing a melody. Also a previous flash could have gone wrong and it needs longer to boot up than usual.
There are simply not more ESCs connected.
The MCU of the ESC is defective.
Try a different USB cable - sometimes they are not properly shielded and might result in such errors.
Make sure your radio is turned off while interacting with the configurator. Switched on radios have been reported to cause issues with electro-magnetic interference and thus failed read/writes.
If the result is not what you are expecting, try clicking the 'Read Settings' button again.

Wasee045, 2024-11-13:
[image]

stylesuxx, 2024-11-13:
So, did you try to read again? Is the ESC powered? Which ESC/FC combo are we talking about?

Wasee045, 2024-11-13:
F7 v3 speedy bee stack
Blheli32 50a esc

stylesuxx, 2024-11-13:
Yeah. Use BLHeli_32 configurator, esc-configurator does not support that.

Good Luck.

closed 2024-11-13 by stylesuxx

Wasee045, 2024-11-13:
Thanks, I will!