stylesuxx/esc-configurator #501: ESC problem

AleksandarRekovic, 2024-12-15:
No crash, cooling vista on the bench..I was just try to connect on esc-conf. to update on bluejay and this happened

2024/12/14 22:57:43: Chrome (131.0.0.0) on Windows
2024/12/14 23:04:11: Plugged in
2024/12/14 23:04:25: Opened serial port
2024/12/14 23:04:25: MultiWii API version received: 1.46.0
2024/12/14 23:04:25: Flight controller info, identifier: BTFL version: 4.5.0
2024/12/14 23:04:25: Running firmware released on: Nov 22 2024 19:39:08
2024/12/14 23:04:25: Board: S405, version: 0
2024/12/14 23:04:25: Unique device ID received - 0x03e00343432470a34333839
2024/12/14 23:06:55: Trying to read 4 ESCs
2024/12/14 23:06:58: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:06:58: MessageNotOkError: Message not OK 
2024/12/14 23:06:58: Failed reading ESC 1
2024/12/14 23:07:01: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:07:01: MessageNotOkError: Message not OK 
2024/12/14 23:07:01: Failed reading ESC 2
2024/12/14 23:07:05: Read ESC 3: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:07:06: Read ESC 4: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:07:06: Done reading ESCs
2024/12/14 23:07:47: Closed port
2024/12/14 23:07:48: MSP command failed: Timed out after 1000ms
2024/12/14 23:08:20: Could not open serial port 
2024/12/14 23:08:20: Error: Could not open serial port 
2024/12/14 23:08:20: Port already in use by another application - try re-connecting
2024/12/14 23:08:21: MSP command failed: Timed out after 1000ms
2024/12/14 23:08:39: Unplugged
2024/12/14 23:08:46: Plugged in
2024/12/14 23:08:52: Opened serial port
2024/12/14 23:08:52: MultiWii API version received: 1.46.0
2024/12/14 23:08:52: Flight controller info, identifier: BTFL version: 4.5.0
2024/12/14 23:08:52: Running firmware released on: Nov 22 2024 19:39:08
2024/12/14 23:08:52: Board: S405, version: 0
2024/12/14 23:08:52: Unique device ID received - 0x03e00343432470a34333839
2024/12/14 23:08:56: Trying to read 4 ESCs
2024/12/14 23:08:57: Read ESC 1: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:08:58: Read ESC 2: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:09:01: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:09:01: MessageNotOkError: Message not OK 
2024/12/14 23:09:01: Failed reading ESC 3
2024/12/14 23:09:04: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:09:04: MessageNotOkError: Message not OK 
2024/12/14 23:09:04: Failed reading ESC 4
2024/12/14 23:09:04: Done reading ESCs
2024/12/14 23:09:41: Closed port
2024/12/14 23:09:56: Opened serial port
2024/12/14 23:09:57: MultiWii API version received: 1.46.0
2024/12/14 23:09:57: Flight controller info, identifier: BTFL version: 4.5.0
2024/12/14 23:09:57: Running firmware released on: Nov 22 2024 19:39:08
2024/12/14 23:09:57: Board: S405, version: 0
2024/12/14 23:09:57: Unique device ID received - 0x03e00343432470a34333839
2024/12/14 23:10:01: Trying to read 4 ESCs
2024/12/14 23:10:02: Read ESC 1: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:10:03: Read ESC 2: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:10:06: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:10:06: MessageNotOkError: Message not OK 
2024/12/14 23:10:06: Failed reading ESC 3
2024/12/14 23:10:09: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:10:09: MessageNotOkError: Message not OK 
2024/12/14 23:10:09: Failed reading ESC 4
2024/12/14 23:10:09: Done reading ESCs
2024/12/14 23:10:36: Trying to read 4 ESCs
2024/12/14 23:10:37: Read ESC 1: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:10:38: Read ESC 2: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:10:41: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:10:41: MessageNotOkError: Message not OK 
2024/12/14 23:10:41: Failed reading ESC 3
2024/12/14 23:10:44: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:10:44: MessageNotOkError: Message not OK 
2024/12/14 23:10:44: Failed reading ESC 4
2024/12/14 23:10:44: Done reading ESCs
2024/12/14 23:10:51: Closed port
2024/12/14 23:10:59: Unplugged
2024/12/14 23:11:14: Plugged in
2024/12/14 23:11:20: Opened serial port
2024/12/14 23:11:20: MultiWii API version received: 1.46.0
2024/12/14 23:11:20: Flight controller info, identifier: BTFL version: 4.5.0
2024/12/14 23:11:20: Running firmware released on: Nov 22 2024 19:39:08
2024/12/14 23:11:20: Board: S405, version: 0
2024/12/14 23:11:20: Unique device ID received - 0x03e00343432470a34333839
2024/12/14 23:11:23: Trying to read 4 ESCs
2024/12/14 23:11:24: Read ESC 1: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:11:25: Read ESC 2: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:11:28: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:11:28: MessageNotOkError: Message not OK 
2024/12/14 23:11:28: Failed reading ESC 3
2024/12/14 23:11:31: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:11:31: MessageNotOkError: Message not OK 
2024/12/14 23:11:31: Failed reading ESC 4
2024/12/14 23:11:31: Done reading ESCs
2024/12/14 23:11:40: Closed port
2024/12/14 23:11:56: Chrome (131.0.0.0) on Windows
2024/12/14 23:11:57: Unplugged
2024/12/14 23:11:58: Plugged in
2024/12/14 23:12:25: Plugged in
2024/12/14 23:12:42: Opened serial port
2024/12/14 23:12:42: MultiWii API version received: 1.46.0
2024/12/14 23:12:42: Flight controller info, identifier: BTFL version: 4.5.0
2024/12/14 23:12:42: Running firmware released on: Nov 22 2024 19:39:08
2024/12/14 23:12:42: Board: S405, version: 0
2024/12/14 23:12:42: Unique device ID received - 0x03e00343432470a34333839
2024/12/14 23:13:04: Closed port
2024/12/14 23:13:05: MSP command failed: Timed out after 1000ms
2024/12/14 23:16:48: Chrome (131.0.0.0) on Windows
2024/12/14 23:16:48: Plugged in
2024/12/14 23:17:00: Opened serial port
2024/12/14 23:17:00: MultiWii API version received: 1.46.0
2024/12/14 23:17:00: Flight controller info, identifier: BTFL version: 4.5.0
2024/12/14 23:17:00: Running firmware released on: Nov 22 2024 19:39:08
2024/12/14 23:17:00: Board: S405, version: 0
2024/12/14 23:17:00: Unique device ID received - 0x03e00343432470a34333839
2024/12/14 23:17:24: Trying to read 4 ESCs
2024/12/14 23:17:26: Read ESC 1: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:17:27: Read ESC 2: J-H-50 - BLHeli_S, 16.7
2024/12/14 23:17:29: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:17:29: MessageNotOkError: Message not OK 
2024/12/14 23:17:29: Failed reading ESC 3
2024/12/14 23:17:32: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/14 23:17:32: MessageNotOkError: Message not OK 
2024/12/14 23:17:32: Failed reading ESC 4
2024/12/14 23:17:32: Done reading ESCs
2024/12/14 23:19:15: Closed port
2024/12/14 23:19:16: MSP command failed: Timed out after 1000ms
2024/12/14 23:19:39: Unplugged
2024/12/14 23:19:45: Plugged in
2024/12/14 23:24:43: Unplugged
2024/12/14 23:24:46: Plugged in
2024/12/14 23:27:59: Unplugged
2024/12/14 23:28:02: Plugged in
2024/12/14 23:29:10: Unplugged
2024/12/14 23:29:13: Plugged in
2024/12/14 23:30:05: Unplugged
2024/12/14 23:30:11: Plugged in
2024/12/14 23:30:14: Unplugged
2024/12/14 23:31:22: Plugged in
2024/12/14 23:31:35: Chrome (131.0.0.0) on Windows
2024/12/14 23:31:35: Plugged in
2024/12/14 23:34:24: Opened serial port
2024/12/14 23:34:24: MultiWii API version received: 1.46.0
2024/12/14 23:34:24: Flight controller info, identifier: BTFL version: 4.5.0
2024/12/14 23:34:24: Running firmware released on: Nov 22 2024 19:39:08
2024/12/14 23:34:24: Board: S405, version: 0
2024/12/14 23:34:24: Unique device ID received - 0x03e00343432470a34333839

stylesuxx, 2025-01-06:
So you can't read ESC3 and 4 if I see correctly?
Your radio is off and you have tried different USB cables?

AleksandarRekovic, 2025-01-06:
Yes, 3 and 4... few cables, few usb ports, two pc with off radio and nothing.. i solved the problem with arduino nano, c2d, c2ck ports and bsheli suiteSent from my iPhone

stylesuxx, 2025-01-06:
So you already recovered via C2 interface - great!
Closing this issue then.