stylesuxx/esc-configurator #502: While updating to Blue Jay 0.21.0, the connection was lost. After that, I tried reading again, but esc is not recognized. What should I do in this case?

mds080130, 2024-12-16:
[esc-configurator-log.txt](https://github.com/user-attachments/files/18156427/esc-configurator-log.txt)

As the title suggests, the connection was lost while updating to Blue Jay 0.21.0. After that, I tried reading again, but esc2 is not recognized. How do I resolve this?

stylesuxx, 2024-12-16:
Leave it plugged in for a minute and try to read again. If that does not work, you will have to recover via Arduino and C2 interface: https://stylesuxx.github.io/arduino-c2-flasher/

WaspFPV, 2024-12-19:
Have you tried force flashing it by enabling the option to Ignore inappropriate MCU and Layout? (make sure you flash the correct layout, check the other ESCs)

flatdeh, 2024-12-22:
Hi, I have the similar issue. I run updating for all ESC to BlueJay A-X-5 0.21.0 48kHZ after finish only ESC1 was updated to 0.21.0 version and than I run updating again and during second updating, process stuck. After reconnect I found out that ESC1 disappeared from the list. Is there any way to restore esc? My FC is Betafpv [F4 1S 5A AIO ](https://betafpv.com/collections/brushless-flight-controller/products/f4-1s-5a-aio-brushless-flight-controller-elrs-2-4g?variant=39968364757126). Logs:
`2024/12/21 22:28:45: Trying to read 4 ESCs
2024/12/21 22:28:45: Read ESC 1: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:28:45: Read ESC 2: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:28:46: Read ESC 3: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:28:46: Read ESC 4: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:28:46: Done reading ESCs
2024/12/21 22:29:01: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/A_X_5_48_v0.21.0.hex 
2024/12/21 22:29:02: Flashing ESC 1
2024/12/21 22:29:02: ### Step 1: Erasing EEPROM and writing safeguards 
2024/12/21 22:29:02: ### Step 2: Writing bootloader failsafe (brick protection) 
2024/12/21 22:29:04: ### Step 3: Erase, write and verify Pages 0x02-0x0D 
2024/12/21 22:29:17: ### Step 4: Erase, write and verify Pages 0x00-0x02 
2024/12/21 22:29:23: ### Step 5: Write EEPROM section 
2024/12/21 22:29:26: Flashed ESC 1 - 24.3s
2024/12/21 22:29:26: Bluejay layout found 
2024/12/21 22:29:26: Migrated setting RPM_POWER_SLOPE 
2024/12/21 22:29:26: Migrated setting MOTOR_DIRECTION 
2024/12/21 22:29:26: Migrated setting BRAKING_STRENGTH 
2024/12/21 22:29:26: Migrated setting COMMUTATION_TIMING 
2024/12/21 22:29:26: Migrated setting BEEP_STRENGTH 
2024/12/21 22:29:26: Migrated setting BEACON_STRENGTH 
2024/12/21 22:29:26: Migrated setting BEACON_DELAY 
2024/12/21 22:29:26: Migrated setting DEMAG_COMPENSATION 
2024/12/21 22:29:26: Migrated setting BRAKE_ON_STOP 
2024/12/21 22:29:26: Migrated setting LED_CONTROL 
2024/12/21 22:29:26: Migrated setting POWER_RATING 
2024/12/21 22:29:26: Migrated setting STARTUP_MELODY 
2024/12/21 22:29:27: Updating ESC 1 - finished
2024/12/21 22:31:09: Trying to read 4 ESCs
2024/12/21 22:31:10: Read ESC 1: A-X-5 - Bluejay, 0.21.0, 48kHz
2024/12/21 22:31:10: Read ESC 2: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:31:10: Read ESC 3: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:31:11: Read ESC 4: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:31:11: Done reading ESCs
2024/12/21 22:31:28: Updating ESC 1 - finished
2024/12/21 22:31:29: Updating ESC 2 - finished
2024/12/21 22:31:29: Updating ESC 3 - finished
2024/12/21 22:31:30: Updating ESC 4 - finished
2024/12/21 22:31:45: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/A_X_5_48_v0.21.0.hex 
2024/12/21 22:31:45: Flashing ESC 1
2024/12/21 22:31:45: ### Step 1: Erasing EEPROM and writing safeguards 
2024/12/21 22:31:45: ### Step 2: Writing bootloader failsafe (brick protection) 
2024/12/21 22:31:47: ### Step 3: Erase, write and verify Pages 0x02-0x0D 
2024/12/21 22:32:00: ### Step 4: Erase, write and verify Pages 0x00-0x02 
2024/12/21 22:32:14: Failed processing command cmd_DeviceWrite after 10 retries. 
2024/12/21 22:34:06: Unplugged
2024/12/21 22:34:06: Reader failed NetworkError: The device has been lost.
2024/12/21 22:34:11: Plugged in
2024/12/21 22:34:20: Chrome (131.0.0.0) on Mac OS
2024/12/21 22:34:20: Plugged in
2024/12/21 22:34:22: Opened serial port
2024/12/21 22:34:22: MultiWii API version received: 1.46.0
2024/12/21 22:34:22: Flight controller info, identifier: BTFL version: 4.5.1
2024/12/21 22:34:22: Running firmware released on: Nov 15 2024 14:36:36
2024/12/21 22:34:22: Board: S411, version: 0
2024/12/21 22:34:22: Unique device ID received - 0x02500633931500220363458
2024/12/21 22:34:25: Trying to read 4 ESCs
2024/12/21 22:34:27: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/21 22:34:27: MessageNotOkError: Message not OK 
2024/12/21 22:34:27: Failed reading ESC 1
2024/12/21 22:34:28: Read ESC 2: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:34:28: Read ESC 3: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:34:28: Read ESC 4: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:34:28: Done reading ESCs
2024/12/21 22:34:53: Updating ESC 2 - finished
2024/12/21 22:34:54: Updating ESC 3 - finished
2024/12/21 22:34:54: Updating ESC 4 - finished
2024/12/21 22:34:54: Trying to read 4 ESCs
2024/12/21 22:34:57: Failed processing command cmd_DeviceInitFlash after 5 retries. 
2024/12/21 22:34:57: MessageNotOkError: Message not OK 
2024/12/21 22:34:57: Failed reading ESC 1
2024/12/21 22:34:57: Read ESC 2: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:34:58: Read ESC 3: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:34:58: Read ESC 4: A-X-5 - Bluejay, 0.19, 48kHz
2024/12/21 22:34:58: Done reading ESCs
2024/12/21 22:35:09: Closed port`

stylesuxx, 2024-12-22:
@flatdeh Yours is different, your does not show up as unknown anymore, so flashing by ignoring MCU layout as @WaspFPV suggested will not work.

You'll probably have to recover via C2 interface then.

flatdeh, 2024-12-22:
@stylesuxx Thank you for your quick response. I will try to recover via C2 interface, do you have any additional info about it? Does arduino 2c flasher work with arduino mega 2560?

stylesuxx, 2025-01-16:
@flatdeh I don't think so, uno and micro are the only ones I have tested. Mega always is a bit "strange"...

flatdeh, 2025-01-17:
@stylesuxx I bought micro and recovered it using https://stylesuxx.github.io/arduino-c2-flasher/  tool. Thank you.

closed 2025-01-17 by stylesuxx