stylesuxx/esc-configurator #506: unable to flash esc 1 and esc 2

MayurM27, 2025-02-19:
2025/02/19 22:57:37: Chosen firmware: https://raw.githubusercontent.com/bitdump/BLHeli/master/BLHeli_S SiLabs/Hex files/G_H_30_REV16_7.HEX 
2025/02/19 22:57:37: Layout revision 208 is not yet supported
2025/02/19 22:57:37: Flashing ESC 1
2025/02/19 22:57:37: Layout mismatch, override not enabled - aborted
2025/02/19 22:57:37: Flashed ESC 1 - 0.1s
2025/02/19 22:57:37: Layout revision 208 is not yet supported

stylesuxx, 2025-02-19:
You need to check the "ignore" MCU layout checkbox... Make sure you are flashing the correct thing though. Usually this checkbox only needs to be checked after a failed flash...

MayurM27, 2025-02-19:
[image]

ignore mcu layout checkbox option not avialble

stylesuxx, 2025-02-19:
It is, in the step where you select the firmware.

MayurM27, 2025-02-19:
ok got it thank you

MayurM27, 2025-02-19:
admin you can delete this post

closed 2025-02-19 by stylesuxx