stylesuxx/esc-configurator #507: Failed Reading ESC

MichaelThamm, 2025-02-26:
_TLDR; I want to be able to flash the factory firmware BLHeli_S J-H-40 back on to BLS 55A ESC with esc-configurator.com_

I have a SpeedyBee F405 V4 x BLS 55A ESC stack which comes pre-installed with firmwares FC (betaflight) and ESC (BLHeli_S J-H-40). Everything was working (I was able to spin the motors via [esc-configurator.com](http://esc-configurator.com). I tried to flash BlueJay (latest) on to my ESC and test the motors spinning.

The BlueJay flash succeeded however, I was unable to spin the motors. The next morning I tried to connect again via [esc-configurator.com](esc-configurator.com) and I was able to spin the motors again, weird ...

Even weirder is that now when I connect to the FC (esc-configurator.com) successfully, the "Read settings" button produces an error log:

> Failed reading ESC

with the following warning message:

[image]

1. The ESC was not ready yet. This can for example happen if it is playing a melody. Also a previous flash could have gone wrong and it needs longer to boot up than usual.
2. The MCU of the ESC is defective.
3. Make sure your radio is turned off while interacting with the configurator. Switched on radios have been reported to cause issues with electro-magnetic interference and thus failed read/writes.

Point #3 is interesting because I have a distance sensor soldered to the FC but its not really emitting radio waves. The only other peripherals connected are my motors.

I am at a loss for what to do and I really hope its not option #1 or #2.

[esc-configurator-log.txt](https://github.com/user-attachments/files/18976021/esc-configurator-log.txt)

stylesuxx, 2025-02-26:
You do have a battery connected, right?

MichaelThamm, 2025-02-27:
> You do have a battery connected, right?
> Also you tried a different cable?
> 
> Very unusual that all ESCs would not show up anymore after successful flash...

Yes I had my drone battery connected and I tried 3 different cables which were all data cables. I also used the same cabke to flash the FC to Ardupilot and back to Betaflight.

Is there any way for me to get BLHeli_S back on the ESC with my issues in esc-configurator.com?

Should I buy a new ESC?

stylesuxx, 2025-02-27:
This all sounds very strange to me...
Do the motors beep when you plug the copter in?

Could you please try re-flashing the initial BF target that the FC came with?

In the worst case you can always recover via C2 interface (you need an Arduino though) - so no need to buy a new ESC. I don't think that this is necessary though. I have never seen 4 ESCs just stop working after a successful flash - especially them coming back to working after they did not. I assume there is something else at play here...

stylesuxx, 2025-03-08:
Hey, any update on this?

MichaelThamm, 2025-03-09:
Ive never had to recover an ESC firmware and will need to research the process. What do you mean by

> Could you please try re-flashing the initial BF target that the FC came with?

The issue is not with the FC firmware its with the ESC. I did flash latest BF stable with betaflight configurator though. Is this what you mean; that I should go back to the BF that it shipped with?

> recover via C2 interface (you need an Arduino though) - so no need to buy a new ESC

This makes me happy to hear but a huge pain. I have an arduino and will likely try this process.

Will close the issue proactively and can always open again if needed.

MichaelThamm, 2025-03-12:
I will recover with the C2 interface on the board and try again. Thanks for the feedback!

closed 2025-03-12 by MichaelThamm

stylesuxx, 2025-03-12:
Let me know how it goes - again, very uncommon to lose all of the ESCs at once.

Also let me know if the motors are beeping when you plug them in - this usually is a good sign that the controllers are not bricked.