stylesuxx/esc-configurator #508: Can not use ESC Configuration in Ardipilot firmware, only can use in BF firmware.

Kumamon92, 2025-03-08:
My FC with Ardupilot 4.7.0 will show "port already in use by another application - try re-connecting" if I connect it with ESC Configuration.  But it works after I flash back to BF 4.5.1. Hope the working group can make configuration support the APM firmware in the future.

stylesuxx, 2025-03-08:
This will most likely be something that needs fixing in Ardupilot.

Since you can connect with BF flashed I assume that you don't have any other software running that could be claiming the serial interface. If you post your log, I can see if there is something obvious.

"Port already in use" usually means that the port is either not claimable or that it is not responding with expected data (the second case, I might be able to fix on my end).

Kumamon92, 2025-03-13:
Thanks for your reply. I watched some video after your advice. I find I need to change some parameter so that the configuration could be used in ardupilot firmware.
The title of the parameter is ”SERVO_BLH_AUTO". Change the number from 0 to 1, the configuration will works.
Thanks again for you reply.

closed 2025-03-13 by stylesuxx