stylesuxx/esc-configurator #510: Cant read 2 esc flash failed

jadowse, 2025-04-03:
I've tried to update my meteor65 pro o4 with bluejay, the flash appeared to work but when i went to read the settings again two esc have failed to read and the motors wont start up.

stylesuxx, 2025-04-03:
Post the log please. Do they show up as "Unknown" or not at all?

jadowse, 2025-04-04:
[esc-configurator-log.txt](https://github.com/user-attachments/files/19596186/esc-configurator-log.txt)
seems to be going bad to worse...

jadowse, 2025-04-04:
Failed reading esc 1,2 and 3

stylesuxx, 2025-04-04:
Such flaky behavior is usually a sign for a bad USB cable. Also your radio is off, right?

Also your log only shows the last read - I'd like to see the rest of the log where I can see what happened during the flash.

jadowse, 2025-04-04:
Radio off. I'll see what I can grab. Cable is brand new and good. Is this
recoverable?

jadowse, 2025-04-04:
[esc-configurator-log.txt](https://github.com/user-attachments/files/19596470/esc-configurator-log.txt) this is all i have as i didnt grab logs after it failed.