stylesuxx/esc-configurator #515: Could not get file for flashing, missing file for ESC: J-L-15

BozoBonobo, 2025-04-21:
Trying to flash my ESC on GEP-TAKER G4 35A AIO, and receiving error "Could not get file for flashing". Try to download hex manually from debug link and seems file is missing as showong github 404 error. Can you please fix issue? 

Trying to flash firmware: 
Firmware: Bluejay
ESC: J-L-15
Version: v0.21.0
PWM Frequency: 48

Debug Log:

>2025/04/21 22:21:42: Edge (135.0.0.0) on Windows
2025/04/21 22:21:45: Plugged in
2025/04/21 22:21:47: Opened serial port
2025/04/21 22:21:47: MultiWii API version received: 1.46.0
2025/04/21 22:21:47: Flight controller info, identifier: BTFL version: 4.5.0
2025/04/21 22:21:47: Running firmware released on: Dec  8 2023 12:04:40
2025/04/21 22:21:47: Board: SG47, version: 0
2025/04/21 22:21:47: Unique device ID received - 0x04300555246500c20333536
2025/04/21 22:22:04: Trying to read 4 ESCs
2025/04/21 22:22:05: Read ESC 1: J-L-15 - BLHeli_S, 16.7
2025/04/21 22:22:06: Read ESC 2: J-L-15 - BLHeli_S, 16.7
2025/04/21 22:22:07: Read ESC 3: J-L-15 - BLHeli_S, 16.7
2025/04/21 22:22:08: Read ESC 4: J-L-15 - BLHeli_S, 16.7
2025/04/21 22:22:08: Done reading ESCs
2025/04/21 22:22:26: Chosen firmware: https://github.com/bird-sanctuary/bluejay/releases/download/v0.21.0/J_L_15_48_v0.21.0.hex 
2025/04/21 22:22:27: Failed fetching firmware 
2025/04/21 22:22:27: Could not get file for flashing.

stylesuxx, 2025-04-21:
L ESCs are no longer supported. User v0.18.1

closed 2025-04-21 by stylesuxx

BozoBonobo, 2025-04-22:
Hello Chris,
Is there way still to upgrade my ESC to bluejay, or my FC is outdated and I
can't do anything?

stylesuxx, 2025-04-22:
You can upgrade to Bluejay, you will just have to stick to version 0.18.1