stylesuxx/esc-configurator #518: Configurator not finding ESCs

luuk00000000000000000000000000000000000, 2025-04-28:
Hello,

I want to change the startup melodies of the ESCs on my betaFPV Air65 (this uses their Air 5in1 controller AFAIK), however it seems that ESC configurator can't find the ESCs at all. It gives the `Failed reading ESC 1|2|3|4` error.

Only sometimes it can find one or more of the ESCs, but I've never gotten it to read all of the ESCs. The motors spin fine and bidirectional DSHOT seems to be working as well.

I've attached my debug logfile, I hope you can help me out!

[esc-configurator-log (6).txt](https://github.com/user-attachments/files/19949678/esc-configurator-log.6.txt)

stylesuxx, 2025-04-28:
Battery is attached, and radio is off?

luuk00000000000000000000000000000000000, 2025-04-28:
Yes, radio is turned off and a battery is attached. That's why I'm finding this to be so strange. VTX is also set to pitmode.

stylesuxx, 2025-04-28:
And you have tried different USB cables?

luuk00000000000000000000000000000000000, 2025-04-28:
Yes, although it didn't seem to make a difference. The Air65 connects through USB via a small adapter, but I don't know if this could be the culprit (seeing that other people have gotten it to work just fine).

luuk00000000000000000000000000000000000, 2025-04-28:
I've set up my ELRS RX to automatically connect to my home WiFi if I don't connect my TX within a minute. I can check tomorrow if this could be the issue.

stylesuxx, 2025-04-28:
Yeah, I have the Air65 myself and am having no issues with it whatsoever.

You say sometimes the ESCs will show up? Can you post a log where that is happening? Any chance you can try on a different machine?

stylesuxx, 2025-04-28:
Yes, please check that and report back.

luuk00000000000000000000000000000000000, 2025-04-29:
I've tried to turn off ELRS RX WiFi, but it didn't seem to help. I've also tried to use another computer to no avail. I _did_ manage to catch the event when it reads one of the ESCs, log attached below.

```
2025/04/29 20:18:32: Plugged in
2025/04/29 20:18:45: Opened serial port
2025/04/29 20:18:45: MultiWii API version received: 1.46.0
2025/04/29 20:18:45: Flight controller info, identifier: BTFL version: 4.5.2
2025/04/29 20:18:45: Running firmware released on: Apr  3 2025 03:18:57
2025/04/29 20:18:45: Board: SG47, version: 0
2025/04/29 20:18:45: Unique device ID received - 0x05000425246501720333536
2025/04/29 20:18:49: TX: cmd_DeviceInitFlash - 0x0 
2025/04/29 20:18:49: Trying to read 4 ESCs
2025/04/29 20:18:49: RX: cmd_DeviceInitFlash - 0xB5,0xE8,0x64,0x1 
2025/04/29 20:18:49: TX: cmd_DeviceRead @ 0x3000 - 0x70 
2025/04/29 20:19:01: Failed processing command cmd_DeviceRead after 10 retries. 
2025/04/29 20:19:01: ESC 1 read settings failed Message not OK MessageNotOkError: Message not OK
2025/04/29 20:19:01: Error: MessageNotOkError: Message not OK 
2025/04/29 20:19:01: TX: cmd_DeviceInitFlash - 0x1 
2025/04/29 20:19:01: Failed reading ESC 1
2025/04/29 20:19:01: RX: cmd_DeviceInitFlash - 0xB5,0xE8,0x64,0x1 
2025/04/29 20:19:01: TX: cmd_DeviceRead @ 0x3000 - 0x70 
2025/04/29 20:19:09: RX: cmd_DeviceRead @ 0x3000 - 0x0,0x13,0xCE,0xFF,0xFF,0x1,0x1,0x32,0xFF,0x9,0x60,0x1,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x3B,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x7,0xFF,0xFF,0xFF,0x0,0x0,0x2,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x23,0x41,0x5F,0x58,0x5F,0x30,0x35,0x23,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x23,0x42,0x4C,0x48,0x45,0x4C,0x49,0x24,0x45,0x46,0x4D,0x38,0x42,0x35,0x31,0x23,0x42,0x6C,0x75,0x65,0x6A,0x61,0x79,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20 
2025/04/29 20:19:09: TX: cmd_DeviceRead @ 0x3000 - 0xFF 
2025/04/29 20:19:22: Failed processing command cmd_DeviceRead after 10 retries. 
2025/04/29 20:19:22: ESC 2 read settings failed Message not OK MessageNotOkError: Message not OK
2025/04/29 20:19:22: Error: MessageNotOkError: Message not OK 
2025/04/29 20:19:22: TX: cmd_DeviceInitFlash - 0x2 
2025/04/29 20:19:22: Failed reading ESC 2
2025/04/29 20:19:22: RX: cmd_DeviceInitFlash - 0xB5,0xE8,0x64,0x1 
2025/04/29 20:19:22: TX: cmd_DeviceRead @ 0x3000 - 0x70 
2025/04/29 20:19:28: RX: cmd_DeviceRead @ 0x3000 - 0x0,0x13,0xCE,0xFF,0x33,0x1,0x1,0x19,0xFF,0x9,0x60,0x1,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x28,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x7,0xFF,0xFF,0xFF,0x0,0x0,0x2,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x23,0x41,0x5F,0x58,0x5F,0x30,0x35,0x23,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x23,0x42,0x4C,0x48,0x45,0x4C,0x49,0x24,0x45,0x46,0x4D,0x38,0x42,0x35,0x31,0x23,0x42,0x6C,0x75,0x65,0x6A,0x61,0x79,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20 
2025/04/29 20:19:28: TX: cmd_DeviceRead @ 0x3000 - 0xFF 
2025/04/29 20:19:30: RX: cmd_DeviceRead @ 0x3000 - 0x0,0x13,0xCE,0xFF,0x33,0x1,0x1,0x19,0xFF,0x9,0x60,0x1,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x28,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x7,0xFF,0xFF,0xFF,0x0,0x0,0x2,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x23,0x41,0x5F,0x58,0x5F,0x30,0x35,0x23,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x23,0x42,0x4C,0x48,0x45,0x4C,0x49,0x24,0x45,0x46,0x4D,0x38,0x42,0x35,0x31,0x23,0x42,0x6C,0x75,0x65,0x6A,0x61,0x79,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x2,0x3A,0x4,0x20,0x34,0x42,0xD,0x0,0x45,0x2D,0xD,0x0,0x34,0x42,0xD,0x0,0x4E,0x27,0xD3,0x0,0x45,0x2D,0xD0,0x19,0x34,0x19,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF 
2025/04/29 20:19:30: TX: cmd_DeviceInitFlash - 0x3 
2025/04/29 20:19:30: Read ESC 3: A-X-5 - Bluejay, 0.19, 96kHz
2025/04/29 20:19:30: RX: cmd_DeviceInitFlash - 0xB5,0xE8,0x64,0x1 
2025/04/29 20:19:30: TX: cmd_DeviceRead @ 0x3000 - 0x70 
2025/04/29 20:19:32: RX: cmd_DeviceRead @ 0x3000 - 0x0,0x13,0xCE,0xFF,0x33,0x1,0x1,0x19,0xFF,0x9,0x60,0x1,0xFF,0x55,0xAA,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x4,0xFF,0xFF,0xFF,0xFF,0xFF,0x28,0x50,0x4,0xFF,0x2,0xFF,0xFF,0xFF,0x7,0xFF,0xFF,0xFF,0x0,0x0,0x2,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x23,0x41,0x5F,0x58,0x5F,0x30,0x35,0x23,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x23,0x42,0x4C,0x48,0x45,0x4C,0x49,0x24,0x45,0x46,0x4D,0x38,0x42,0x35,0x31,0x23,0x42,0x6C,0x75,0x65,0x6A,0x61,0x79,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20,0x20 
2025/04/29 20:19:32: TX: cmd_DeviceRead @ 0x3000 - 0xFF 
2025/04/29 20:19:45: Failed processing command cmd_DeviceRead after 10 retries. 
2025/04/29 20:19:45: ESC 4 read settings failed Message not OK MessageNotOkError: Message not OK
2025/04/29 20:19:45: Error: MessageNotOkError: Message not OK 
2025/04/29 20:19:45: Failed reading ESC 4
2025/04/29 20:19:45: Done reading ESCs
```

luuk00000000000000000000000000000000000, 2025-04-29:
I flashed the factory firmware back to the stock BetaFPV image, and it now works fine. I don't know what they configured differently in the stock image though.

stylesuxx, 2025-04-30:
Interesting, so only after flashing with the latest betaflight release it does not work?
Could you tell me which BF version the stock image has? Also maybe a settings dump?

I'm not going to be able to test anything until next week myself, but I'd be interested in trying to reproduce this (hoping it's not a general issue with the latest BF release, but I think then more people would complain).

closed 2026-11-10 by stylesuxx