stylesuxx/esc-configurator #520: the problem with the firmware

maxmedoffuzb, 2025-06-04:
I can't flash esc from betafpv air 65 .

[esc-configurator-log (1).txt](https://github.com/user-attachments/files/20599986/esc-configurator-log.1.txt)
[esc-configurator-log (2).txt](https://github.com/user-attachments/files/20599985/esc-configurator-log.2.txt)
[esc-configurator-log.txt](https://github.com/user-attachments/files/20599984/esc-configurator-log.txt)

stylesuxx, 2025-06-04:
Which log should I be looking at, what's your exact issue? Seems like a failed flash and you don't have "Ignore MCU" layout enabled. Please provide more info.

maxmedoffuzb, 2025-06-10:
Thank you very much stylesuxx, your advice helped.It turns out I didn't check the box "ignore mcu".I flashed on bluejay 0.21 96 khz. The site provides firmware Z-H-30 (blheli), but for some reason I can't flash this firmware.Bluejay is well stitched.Thanks again for your help.
[image]