stylesuxx/esc-configurator #523: Smoked motor after flashing 24khz and changing start-up theme

Bingoliath, 2025-06-15:
Hi,

Before I used the most recent version of bluejay .21, I believe.

I reflashed the ESCs to use 24Khz PWM frequency and set the Doom theme to play on startup.

As soon as the Doom theme started, I smoked my motor.

In this case, I would scream that it was either the ESC or the Motor itself, however, I’m not sure, because the default theme worked and my motor smoked as i changed the PWM frequency and the theme.

Maybe it’s a software issue?

stylesuxx, 2025-06-16:
Don't hook up a new motor to this ESC. If the ESC is faulty, you are running risk to smoke that motor too.

I was assuming you flashed all ESCs of your copter, so I was wondering if only one of those motors smoked. With default settings that should definitely not happen as long as motor and ESC are in good condition.

Bingoliath, 2025-06-16:
Yes, one of the FETs on ESC 2 were shorted. I had continuity between the positive and pad number 3.

However, currently I can’t say what happened. The ESC was new - had two flights on it. Previously flashed with .21 and 48Khz.

What I can say is that this happened when I flashed 24KHz and was playing around with the start up chimes. I actually tried the Windows one before the Doom one and it was fine.

Likely I’m guessing this is related to a QC issue on the speedybee or maybe the start up chimes was too loud on the default loudness. Can’t point any fingers 🫠

Thank you for responding. Closing it.