stylesuxx/esc-configurator #524: Unable to Read ESC Settings in ESC Configurator

KarthiAru, 2025-07-08:
I'm currently unable to read the ESC settings using the ESC Configurator. When I click "Read Settings", nothing happens, and the browser console shows some errors. I'm not sure if these errors are related, but the issue persists across both Chrome and Edge (latest versions). It was working fine a few days ago, so it's possible that a recent website update introduced the issue.

[image]

For context, I'm able to read the ESC settings using the BLHeli and Bluejay standalone configurators without any problems. However, I need to flash the latest Bluejay firmware, which only seems to work reliably through the ESC Configurator—flashing via the Bluejay Configurator fails.

[image]

[image]

KarthiAru, 2025-07-08:
Debug Log:

```
2025/07/08 17:18:50: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:18:51: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:18:52: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:18:53: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:18:54: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:18:54: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:19:16: Chrome (138.0.0.0) on Windows
2025/07/08 17:19:16: Plugged in
2025/07/08 17:23:22: Unplugged
2025/07/08 17:23:22: Unplugged
2025/07/08 17:23:28: Plugged in
2025/07/08 17:23:28: Plugged in
2025/07/08 17:23:43: Chrome (138.0.0.0) on Windows
2025/07/08 17:23:43: Plugged in
2025/07/08 17:23:46: Opened serial port
2025/07/08 17:23:46: MultiWii API version received: 1.42.0
2025/07/08 17:23:46: Flight controller info, identifier: ARDU version: 3.3.1
2025/07/08 17:23:46: Running firmware released on: Mar 16 2018 08:42:29
2025/07/08 17:23:46: Board: ARCH, version: 0
2025/07/08 17:23:46: Unique device ID received - 0x02f00243532511636373732
2025/07/08 17:23:47: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:23:48: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:23:49: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:26:12: Chrome (138.0.0.0) on Windows
2025/07/08 17:26:12: Plugged in
2025/07/08 17:28:14: Chrome (138.0.0.0) on Windows
2025/07/08 17:28:14: Plugged in
2025/07/08 17:28:18: Port changed
2025/07/08 17:28:30: Port selected
2025/07/08 17:28:31: Opened serial port
2025/07/08 17:28:31: MultiWii API version received: 1.42.0
2025/07/08 17:28:31: Flight controller info, identifier: ARDU version: 3.3.1
2025/07/08 17:28:31: Running firmware released on: Mar 16 2018 08:42:29
2025/07/08 17:28:31: Board: ARCH, version: 0
2025/07/08 17:28:31: Unique device ID received - 0x02f00243532511636373732
2025/07/08 17:28:32: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:28:33: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:28:34: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:29:02: Chrome (138.0.0.0) on Windows
2025/07/08 17:29:02: Plugged in
2025/07/08 17:30:27: Chrome (138.0.0.0) on Windows
2025/07/08 17:40:50: Chrome (138.0.0.0) on Windows
2025/07/08 17:40:50: Plugged in
2025/07/08 17:41:04: Opened serial port
2025/07/08 17:41:04: MultiWii API version received: 1.42.0
2025/07/08 17:41:04: Flight controller info, identifier: ARDU version: 3.3.1
2025/07/08 17:41:04: Running firmware released on: Mar 16 2018 08:42:29
2025/07/08 17:41:04: Board: ARCH, version: 0
2025/07/08 17:41:04: Unique device ID received - 0x02f00243532511636373732
2025/07/08 17:41:05: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:41:06: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:41:06: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:41:20: Chrome (138.0.0.0) on Windows
2025/07/08 17:41:20: Plugged in
2025/07/08 17:41:23: Opened serial port
2025/07/08 17:41:23: MultiWii API version received: 1.42.0
2025/07/08 17:41:23: Flight controller info, identifier: ARDU version: 3.3.1
2025/07/08 17:41:23: Running firmware released on: Mar 16 2018 08:42:29
2025/07/08 17:41:23: Board: ARCH, version: 0
2025/07/08 17:41:23: Unique device ID received - 0x02f00243532511636373732
2025/07/08 17:41:24: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:41:25: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 17:44:23: Chrome (138.0.0.0) on Windows
2025/07/08 17:44:23: Plugged in
2025/07/08 17:45:45: Unplugged
2025/07/08 17:45:45: Unplugged
2025/07/08 17:46:02: Plugged in
[... 66 lines cut ...]
2025/07/08 18:40:05: Chrome (138.0.0.0) on Windows
2025/07/08 18:40:10: NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user. 
2025/07/08 18:40:16: Port selected
2025/07/08 18:40:18: Opened serial port
2025/07/08 18:40:18: MultiWii API version received: 1.42.0
2025/07/08 18:40:18: Flight controller info, identifier: ARDU version: 3.3.1
2025/07/08 18:40:18: Running firmware released on: Mar 16 2018 08:42:29
2025/07/08 18:40:18: Board: ARCH, version: 0
2025/07/08 18:40:18: Unique device ID received - 0x02f00243532511636373732
2025/07/08 18:40:19: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 18:40:20: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 18:40:20: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 18:40:41: Chrome (138.0.0.0) on Windows
2025/07/08 18:40:42: Plugged in
2025/07/08 18:40:46: Opened serial port
2025/07/08 18:40:46: MultiWii API version received: 1.42.0
2025/07/08 18:40:46: Flight controller info, identifier: ARDU version: 3.3.1
2025/07/08 18:40:46: Running firmware released on: Mar 16 2018 08:42:29
2025/07/08 18:40:46: Board: ARCH, version: 0
2025/07/08 18:40:46: Unique device ID received - 0x02f00243532511636373732
2025/07/08 18:40:47: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 18:40:48: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 18:40:49: MSP command failed: Offset is outside the bounds of the DataView
2025/07/08 18:40:49: MSP command failed: Offset is outside the bounds of the DataView
```

KarthiAru, 2025-07-08:
Ok. I'm using a Pixhawk 6X based flight controller. I've used PX4 Autopilot and Ardupilot before. I'll need to check if Betaflight is supported on this FC.

KarthiAru, 2025-07-11:
@stylesuxx Yes, I can help with the testing

Swagnerovski, 2025-07-30:
Hello! I am also trying to flash bluejay on my BL Heli ESCs while using a Pixhawk 6X with Ardupilot and I am facing the same Problem as #508 . Kumamon92 was able to flash his ESC with the 4.5.1 version, so I tried using the older version but this wasn´t working.

As buying another FC which is supported by Betaflight seems a bit excessive, has anyone tried using something to directly to the PC like this:
https://www.flyingtech.co.uk/product/esc-usb-linker-programmer-flash-tool-for-blheli-bl-s-bl-32/

alexsys-, 2025-08-16:
I tried 2 different stacks 
speedy bee f405 + 60A bluejay esc
micoair h743 + 50A bluejay esc
in both scenarios I can connect to the esc but cannot read any settings using https://esc-configurator.com/ when being on ardupilot
as soon as I flash betaflight everything works perfectly
I tried all the possible settings I could find online:
including but not limited to 
SERVO_BLH_AUTO=1 
SERVO_BLH_MASK to 15
SERVO_DSHOT_ESC=1 
I also tried to map different servos in ardupilot and nothing
Yes betaflight is a workaround, but when you use ardupilot and experimenting on esc setting is a huge PIA to keep reflashing and re-setting everuthing on every iteration.
If you need to test and experiment on a fix I am willing to help.
BlHeliSuite(not 32) connect but doesnt recognize bluejay firmware
BlHeliSuite32 also connects but it is not a BlHeli32 so it wont work