stylesuxx/esc-configurator #534: Error: Could not get file for flashing

BirdsEyeViewHamburg, 2025-09-06:
Tried to flash a GEPRC Taker F4 35A AIO with BlueJay. ESC Configurator always said: "Could not get file for flashing" Battery was connected to AIO. USB cable worked with other AIOs. Manufacturer says it is compatible with BlueJay.

stylesuxx, 2025-09-07:
Ok, that definitely should not work. Those are different MCUs or Geprc is mislabeling their Firmware - proceed with caution.

Firmware is not named the way it is for fun. Flashing wrong firmware can easily result in fatal problems.

BirdsEyeViewHamburg, 2025-09-07:
Well, it worked. The quad is flying fine, it seems. GEPRC says on the Website for this AIO "Target: J-H-15". Before BLHeli_S was flashed.

stylesuxx, 2025-09-07:
Right...

closed 2025-09-07 by stylesuxx